biemond/solaredge.modbus #78: Growatt : 'Export to grid' information missing ?

MiosDK, 2024-05-21:
Hi, 
I have a Growatt 3600 hybrid inverter where I need the information regarding: 'export to grid' is it possible to this information? 
[image]

biemond, 2024-05-21:
hey, what does this say on yours

```
    "measure_power.gridoutput": {
      "decimals": 0,
      "title": {
        "en": "Grid output",
        "de": "Netzexport"
      }
    },
```

        "gridOutputPower": [40, 2, 'UINT32', "Grid Output Power", -1],

biemond, 2024-05-21:
or do you mean total of today export.  

can you find out what register I need to use for that.?

MiosDK, 2024-05-21:
> hey, what does this say on yours
> 
> ```
>     "measure_power.gridoutput": {
>       "decimals": 0,
>       "title": {
>         "en": "Grid output",
>         "de": "Netzexport"
>       }
>     },
> ```
> 
> ```
>     "gridOutputPower": [40, 2, 'UINT32', "Grid Output Power", -1],
> ```

I dont know to see that ?

MiosDK, 2024-05-21:
> or do you mean total of today export.
> 
> can you find out what register I need to use for that.?

I need the actual export 'up front' not all Day, I should use it regarding, do you know what I Mean? I dont have a register overview, do you know where I can find the register ?

biemond, 2024-05-21:
you should have grid output on your device , what device did you choose?

[image]

MiosDK, 2024-05-21:
> you should have grid output on your device , what device did you choose?
> 
> <img alt="Screen Shot 2024-05-21 at 21 44 08" width="1029" src="https://private-user-images.githubusercontent.com/1085004/332561138-f3bcc085-2ec7-461e-a331-4a54948fc6ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzMjExMzYsIm5iZiI6MTcxNjMyMDgzNiwicGF0aCI6Ii8xMDg1MDA0LzMzMjU2MTEzOC1mM2JjYzA4NS0yZWM3LTQ2MWUtYTMzMS00YTU0OTQ4ZmM2ZWUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjFUMTk0NzE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGVkNzk3YTYzNTJiOWU0MDk4ZGQ0NjQ1ZTA1ZGU5MjFkNzIxM2E2YjA0ZWMwOTQ2MGZjNmZiMjVhZjE4NTE1OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.qppk1DR-cL6Q3AFNSFvsNazGqHk_iL6ulWU1sth_yZI">

This one at the picture, is this the right one ? 

[image]

MiosDK, 2024-05-21:
> > you should have grid output on your device , what device did you choose?
> > <img alt="Screen Shot 2024-05-21 at 21 44 08" width="1029" src="https://private-user-images.githubusercontent.com/1085004/332561138-f3bcc085-2ec7-461e-a331-4a54948fc6ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzMjExMzYsIm5iZiI6MTcxNjMyMDgzNiwicGF0aCI6Ii8xMDg1MDA0LzMzMjU2MTEzOC1mM2JjYzA4NS0yZWM3LTQ2MWUtYTMzMS00YTU0OTQ4ZmM2ZWUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjFUMTk0NzE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGVkNzk3YTYzNTJiOWU0MDk4ZGQ0NjQ1ZTA1ZGU5MjFkNzIxM2E2YjA0ZWMwOTQ2MGZjNmZiMjVhZjE4NTE1OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.qppk1DR-cL6Q3AFNSFvsNazGqHk_iL6ulWU1sth_yZI">
> 
> This one at the picture, is this the right one ?
> 
> ![Screenshot_20240521_214929_Homey~2](https://private-user-images.githubusercontent.com/170458604/332562815-c8051dda-65e9-4b2a-b6b6-92607d6b6dd4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzMjEzNjEsIm5iZiI6MTcxNjMyMTA2MSwicGF0aCI6Ii8xNzA0NTg2MDQvMzMyNTYyODE1LWM4MDUxZGRhLTY1ZTktNGIyYS1iNmI2LTkyNjA3ZDZiNmRkNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyMVQxOTUxMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYTg4NWQ3MGZjZDI0ZTg3MDZhZWY1NjA1NjJhMGNkMzAzYjQwYjJiZDRiZWU3MDRlZTg0N2ViNzRmZTZhMjc3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ecJN3_slElidJDvHinANtmVOSfXTvXItsk13u52-dmE)

I need the actual 'export to grid' readings, so when the battery is 100% and the sun panels is produceing ex. 1200watt and the house is using 600watt the last 600 watt is to be exported to grid, therefor I will start something in the house that will use the the power insted.

biemond, 2024-05-21:
I think you can use the TL one.  and you send me a diagnostic report.  but grid output is the 1

MiosDK, 2024-05-22:
> I think you can use the TL one. and you send me a diagnostic report. but grid output is the 1

TL is also ok can I see, but there is not more options to this matter, I have tried grid output, its not the one to use in this matter, the name should be 'export to grid' can upload a modbus PDF maybe I can find the correct register ?

MiosDK, 2024-05-22:
> > I think you can use the TL one. and you send me a diagnostic report. but grid output is the 1
> 
> TL is also ok can I see, but there is not more options to this matter, I have tried grid output, its not the one to use in this matter, the name should be 'export to grid' can upload a modbus PDF maybe I can find the correct register ? I can see there is a 'Grid export YIELD' thats the total for the Day, I need this information when it change immediately, do you know what meen ? 😃

biemond, 2024-05-22:
when it is not there , I need to calculate it or find the right register.   but you can also do this yourself with solar/batt output minus the load. 

what is not ok on grid output?

MiosDK, 2024-05-22:
> when it is not there , I need to calculate it or find the right register. but you can also do this yourself with solar/batt output minus the load.
> 
> what is not ok on grid output?

I got some information that the modbus registre is 122 + 123 can you fix it ?

biemond, 2024-05-22:
will check those.

MiosDK, 2024-05-22:
> will check those.

New picture from the register here : 
[image]

biemond, 2024-05-22:
looks like it I have already have it on the device or as flow actions , else I will enable it https://github.com/biemond/solaredge.modbus/blob/main/drivers/growattwithbatt_tl/device.ts#L80

https://github.com/biemond/solaredge.modbus/blob/main/drivers/growattwithbatt_tl/device.ts#L28

MiosDK, 2024-05-22:
> oh you want a flow trigger action and block total export.
> 
> currently I support these periods with grid, load or battery priority https://github.com/biemond/solaredge.modbus/blob/main/drivers/growattwithbatt_tl/driver.flow.compose.json

I dont want to Block export but if I let ex. 500watt in export I will start a electric healing element to my heat exchanger to use the export watt...

MiosDK, 2024-05-22:
> looks like it I have already have it on the device or as flow actions , else I will enable it https://github.com/biemond/solaredge.modbus/blob/main/drivers/growattwithbatt_tl/device.ts#L80
> 
> https://github.com/biemond/solaredge.modbus/blob/main/drivers/growattwithbatt_tl/device.ts#L28

Is it the same as the picture ? 
[image]

MiosDK, 2024-05-22:
I dont think the function Is in the flowcard

biemond, 2024-05-22:
you are asking with this 122,123 to block export.   

current grid output should be ok for that.  can you upload the pic on github, not replying on the email.  
also send me a diagnostic report.

MiosDK, 2024-05-22:
> you are asking with this 122,123 to block export.
> 
> current grid output should be ok for that. can you upload the pic on github, not replying on the email. also send me a diagnostic report.

The "only" thing I want, is to see the output mesuring of the 'export to grid' so when its in +xxx watt to grid I want to start my electric device to use the xxx watt in the house and when the export of xxx watt is lens ex. 200watt I want to stop my device to use energy in the house.

I dont know how to send a diagnostic report?

MiosDK, 2024-05-22:
> I got the developer version but like this on my app
> 
> <img alt="Screenshot 2024-05-22 at 14 56 50" width="1752" src="https://private-user-images.githubusercontent.com/1085004/332804543-b67ff8d2-67ac-4f46-b98b-5d9fd72d6cec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzOTAyOTksIm5iZiI6MTcxNjM4OTk5OSwicGF0aCI6Ii8xMDg1MDA0LzMzMjgwNDU0My1iNjdmZjhkMi02N2FjLTRmNDYtYjk4Yi01ZDlmZDcyZDZjZWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjJUMTQ1OTU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmMwYTY5ZDVmZjYzYzY1NDY0ZTUwODJjY2VkZjMzODI1MmQxZTlkZDI4NzhjZDlhMjQ1YzkwZjYwNGU0MDA3YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pizoCmTjq2EG6KIrYzeFLFgAMS0gmjS_uwixOm4EWd8">

SMPL ?

biemond, 2024-05-22:
yeah select my app and press Create Diagonistic report button

MiosDK, 2024-05-22:
> yeah select my app and press Create Diagonistic report button

Ahh Done, I have called it MiosDK-2024

MiosDK, 2024-05-22:
Do you think its possible eith this as I wrote before ? I think the modbus register I wrote before isnt the right one ?

"The "only" thing I want, is to see the output mesuring of the 'export to grid' so when its in +xxx watt to grid I want to start my electric device to use the xxx watt in the house and when the export of xxx watt is lens ex. 200watt I want to stop my device to use energy in the house."

MiosDK, 2024-05-22:
I have this Programmed in the the Growatt inverter - but I want this in Homey instead - Maybe it can help you in this matter and you can find it in the Modbus registre :

[image]

biemond, 2024-05-22:
got your report

biemond, 2024-05-22:
```
l1_current 99 -1 L1 Current
l2_current 0 -1 L2 Current
l3_current 0 -1 L3 Current
temperature 472 -1 Temperature
status 5 0 Status
inputPower 24300 -1 Input Power
outputPower 23099 -1 Output Power
pv1Voltage 2641 -1 pv1 Voltage
pv2Voltage 1811 -1 pv2 Voltage
gridFrequency 4999 -2 Grid Frequency
gridVoltage 2409 -1 Grid Voltage
gridOutputCurrent 98 -1 Grid Output Current
gridOutputPower 23252 -1 Grid Output Power
todayEnergy 224 -1 Today Energy
totalEnergy 111805 -1 Total Energy
pv1InputPower 23263 -1 pv1 Power
pv2InputPower 1109 -1 pv2 Power
pv1TodayEnergy 162 -1 pv2 Today Energy
pv1TotalEnergy 71547 -1 pv2 Total Energy
pv2TodayEnergy 111 -1 pv2 Today Energy
pv2TotalEnergy 39290 -1 pv2 Total Energy
pvEnergyTotal 111063 -1 pv Total Energy
error 0 0 Error
battDischarge 0 -1 battery Discharge
battCharge 0 -1 battery Charge
battvoltage 544 -1 battery Voltage
battsoc 100 0 battery soc
batttemperature 25 -1 battery Temperature
bmssoc 100 0 bms soc
bmstemperature 25 -1 bms Temperature
bmscyclecount 571 0 bms cycle count
bmshealth 94 0 bms soh
bmsstatus 289 0 bms status
bmserror 0 0 bms error
totalhouseload 10000 -1 Total house Load
priority 0 0 priority
today_grid_import 25 -1 Today's Grid Import
total_grid_import 117827 -1 Total Grid Import
today_grid_export 127 -1 Today's Grid Export
total_grid_export 33160 -1 Total Grid Export
today_battery_output_energy 11 -1 Today's Battery Output Energy
total_battery_output_energy 32254 -1 Total Battery Output Energy
today_battery_input_energy 43 -1 Today's Battery Input Energy
total_battery_intput_energy 28064 -1 Total Battery Input Energy
today_load 140 -1 Today's Load
total_load 199920 -1 Total Load
exportlimitenabled 0 0 Export Limit enable
exportlimitpowerrate 0 -1 Export Limit Power Rate
prioritychange 0 0 Priority
gridfirststopsoc 10 0 GridFirst stop SOC
batfirststopsoc 95 0 BatFirst stop SOC
acchargeswitch 0 0 Batt AC charge switch
gridfirststarttime1 4352 0 Grid First Start Time
gridfirststoptime1 5376 0 Grid First Stop Time
gridfirststopswitch1 0 0 Grid First Stop Switch 1
battfirststarttime1 0 0 Battery First Start Time
battfirststoptime1 0 0 Battery First Stop Time
battfirststopswitch1 1 0 Battery First Stop Switch 1
loadfirststarttime1 0 0 Load First Start Time
loadfirststoptime1 0 0 Load First Stop Time
loadfirststopswitch1 0 0 Load First Stop Switch 1
gridfirst1 from: 17:00 ~ 21:00 state: 0
battfirst1 from: 00:00 ~ 00:00 state: 1
```

biemond, 2024-05-22:
so this is your power

your export is gridOutputPower 2325 W or not.
solar = 2430 

```
inputPower 24300 -1 Input Power
outputPower 23099 -1 Output Power
gridOutputPower 23252 -1 Grid Output Power
pv1InputPower 23263 -1 pv1 Power
pv2InputPower 1109 -1 pv2 Power
battDischarge 0 -1 battery Discharge
battCharge 0 -1 battery Charge
```

MiosDK, 2024-05-22:
> so this is your power
>
> your export is gridOutputPower 2325 W or not. solar = 2430
>
> ```
> inputPower 24300 -1 Input Power
> outputPower 23099 -1 Output Power
> gridOutputPower 23252 -1 Grid Output Power [...]
New report send - MiosDK-2024 - 2 - a19b31e1-2393-48d2-b3e9-8e3a4a4b4c54

[image]

biemond, 2024-05-22:
so you can use grid power, but always do 5 min average else it will be on / off every few seconds

MiosDK, 2024-05-22:
> so you can use grid power, but always do 5 min average else it will be on / off every few seconds

I dont think that Gridpower is the solution for this, right now I do not export anything because the sun isnt there any more, I have not at view hvor there is 0 KW - right now GRID output is 931W that is what the house is using (the same as Houseload ) and have nothing with export to Grid - is it possible there is another modbus protocol that can be used ?

biemond, 2024-05-22:
gridOutputPower  should be and it is working for others , not grid power, that is now your batt discharge

MiosDK, 2024-05-22:
> gridOutputPower

I can not see gridOutputPower  anywhere in the app - I have tried to find gridOutputPower  text in both Growatt ?

biemond, 2024-05-22:
it is this 1 ```    "measure_power.gridoutput": {
      "decimals": 0,
      "title": {
        "en": "Grid output",
        "de": "Netzexport"
      }
    },```

MiosDK, 2024-05-23:
> it is this 1 ` "measure_power.gridoutput": { "decimals": 0, "title": { "en": "Grid output", "de": "Netzexport" } },`

Ok, but right now I do not export anything but using power from grid (see pic) and that is what Grid output is showing and not export from sun to grid ?
[image]

biemond, 2024-05-23:
yeah, I see it.  more as load.  It works on someone else as expected. looks your meters are not installed correctly , let check on someone else also.

biemond, 2024-05-23:
[image]

MiosDK, 2024-05-23:
Is it possible there is a difference in witch country the Growatt os placed ? I have the SPH3600 hybrid inverter and a summation meter from electric vendor + smartmeter from Growatt. Do you know if there is others from Denmark there have this solution ? As you can see on this picture, it seems that the import is 0,5 kw, in Homey I think 0,5 kw= bat power + import from grid. Maybe the summation meter is doing something in this matter ? 
The weird thing in this matter that I can do a programming in Growatt app. To do something when sun is exporting to grid with XX watt and shopping again depent how lens Watt is, I think the information is the same in the modbus ? 

[image]

biemond, 2024-05-23:
solar power is pv - batt charge or batt discharge

batt discharge is your solar/batt . 
grid output is + for I think it is import.   
load is both combined here

can you check insights if grid output goes negative with a load of sun and a full batt.   I think negative is export.

MiosDK, 2024-05-23:
What about this mesurering on L1 / Grid output current is this in watt or kw, is it possible that this one is mesuring the output to Grid ? 
[image]

biemond, 2024-05-23:
can you show me your insights graphs for the last 7 days  about grid output. I think sometimes it goes negative . which means export, pos = import

MiosDK, 2024-05-23:
> solar power is pv - batt charge or batt discharge
> 
> batt discharge is your solar/batt . grid output is + for I think it is import. load is both combined here
> 
> can you check insights if grid output goes negative with a load of sun and a full batt. I think negative is export.

The grid is not going negative as you can see at the pic

[image]

biemond, 2024-05-23:
no clue, I just pass it on , let me vpn in to some growatt and see if I see the same.

MiosDK, 2024-05-23:
> no clue, I just pass it on , let me vpn in to some growatt and see if I see the same.

Ok perfect, let me know if I can do.something more in this matter 👍

biemond, 2024-05-23:
you are on the latest firmware or not?

MiosDK, 2024-05-23:
> you are on the latest firmware or not?

I dont now the latest firmware number is ? Maybe you know ? And how can I update if my is a old version ? 

[image]

biemond, 2024-05-23:
oh you have a sph , make sure you don't use tl device but the normal growatt batt device.

MiosDK, 2024-05-23:
> oh you have a sph , make sure you don't use tl device but the normal growatt batt device.

Im using this (see pic) what about the firmware is this the newest ? 

[image]

MiosDK, 2024-05-24:
What about my firmware, is this the newest one ?

biemond, 2024-05-24:
no clue.

biemond, 2024-05-24:
got some for sph, let me add these and then you can send me diagnostic report

[image]

biemond, 2024-05-24:
can you test this https://homey.app/a/solaredge.modbus/test/

and send me a report after 5 min or so. like to see the data.

MiosDK, 2024-05-24:
> can you test this https://homey.app/a/solaredge.modbus/test/
> 
> and send me a report after 5 min or so. like to see the data.

Do I need to uninstall the other version first ?

biemond, 2024-05-24:
no
just install

MiosDK, 2024-05-24:
> no just install

New report sent : d81c15ef-bd0b-4a67-8a46-9aef2ad80ff0

biemond, 2024-05-24:
```
inputPower 0 -1 Input Power
outputPower 9000 -1 Output Power
gridOutputPower 9059 -1 Grid Output Power
pv1InputPower 11 -1 pv1 Power
pv2InputPower 7 -1 pv2 Power
battDischarge 9900 -1 battery Discharge
battCharge 0 -1 battery Charge
totalhouseload 9900 -1 Total house Load
pactouserr 0 -1 import from grid to user
pactousertotal 0 -1 import from grid to user total
pactogrid 0 -1 export to grid
pactogridtotal 683 -1 export to grid total
```

biemond, 2024-05-24:
now we need to check some time where we have import and some export.

biemond, 2024-05-24:
send me a report when you think this is the case

MiosDK, 2024-05-24:
Report is sent with Import : c0dd83ea-583f-4d20-9969-da1a968ab081

[image]

biemond, 2024-05-24:
```
inputPower 0 -1 Input Power
outputPower 0 -1 Output Power
gridOutputPower 0 -1 Grid Output Power
pv1InputPower 1 -1 pv1 Power
pv2InputPower 1 -1 pv2 Power
battDischarge 0 -1 battery Discharge
battCharge 0 -1 battery Charge
pactouserr 54750 -1 import from grid to user
pactousertotal 54617 -1 import from grid to user total
pactogrid 0 -1 export to grid
pactogridtotal 0 -1 export to grid total
```

[17 more replies]