biemond/solaredge.modbus #79: Guidance for first time users

Swamulf, 2024-05-24:
Hi, thank you for providing such great app. I have had an Huawei system for a while and used another app in the Homey store (using web login) but it doesn't work anymore and does not seem to be updated. I would really like to use this one but am unsure of what I need to do. I have entered the IP when trying to add the device. I am however unsure about Port and Device ID. Where do i find those values? Do i need to change any settings on the inverter? I have seen modbus being mentioned but don't know what this is. 

I am thankful for any help on this!

Samuel

Swamulf, 2024-05-24:
Hi, yes I have a dongle which is connected wireless. Do I need to connect via cable?

biemond, 2024-05-24:
can. use the. ip of s dongle port 502 modbus id 1