biemond/solaredge.modbus #80: 2e SolarEdge omvormer koppelen lukt niet.

Davezelf, 2024-05-24:
Beste Edwin,

Ik heb 2 SolarEdge omvormers. 1 daarvan is me gelukt om te koppelen aan Homey met de SE Homey app. Dit heb ik gedaan door de Modbus aan te zetten in de omvormer en daar een poortnummer te geven. (met SetApp van SE) In Homey was de rest een fluitje. Echter de 2e omvormer, wat ik ook probeer, ander IP-adres, ander poortnummer, ik kom er niet mee weg.
Kan jij mij verlichten in hetgeen ik fout doe/vergeet?

biemond, 2024-05-24:
Hey , heeft de 2de omvormer zijn eigen netwerk of is hij gekoppeld aan de 1ste .  dan moet je via 1ste inverter gaan en waarschijnlijk modbus id is 2 dan.

er was een firmware waar modbus over wifi niet werkte, misschien heb je een update nodig.

Davezelf, 2024-05-24:
Hi Edwin,

Dank voor de snelle reactie.

De omvormer van het dak heeft IP 192.168.0.5 poort 1502 en ID 1, is met een ethernetkabel aangesloten, werkt geheel goed.
De omvormer van het schuurdak heeft IP 192.168.0.6 poort 1503 en ID 2. Deze is met een ethernetkabel aangesloten via ethernet over AC. Ze zitten dus op het zelfde netwerk. Ik weet eigenlijk niet of ze gekoppeld zijn. Beide zijn actief en te lezen in de portal. Eergisteren nog de laatste firmware op de omvormers gezet.

Met vriendelijke groet,
Dave

Davezelf, 2024-05-24:
Edwin,

Heb m inmiddels al. De 2e omvormer had ik ID 2 gegeven, moest dus 1 zijn.

Dank voor het meedenken, je bent geweldig.