biemond/solaredge.modbus #81: Solinteg - wrong values

mati686, 2024-05-28:
Hi
I added the Solinteg MHT-10K-25 to my app yesterday. But I have a problem with the data in the application, it seems that some values are different by an order of magnitude.

Solar power is the same as in cloud app.
Battery discharge is same as in cloud app.
Grid output is -6W in homey / 84W in cloud.
PV1 input is 170W in homey / 1,66 kW in cloud.

[image]

biemond, 2024-05-28:
hey, 

looks good to me , your load is 1.8 and that comes from solar and batt discharge.  Grid output is so small diff and this in a moment of time.  

so I think PV1 needs to be 1700W instead.  let me check the docs.  solinteg is a wattsonic I think.

mati686, 2024-05-28:
Yes i think so, PV1 should be 1700W and Grid -60W.
What i read from modbus:
[image]
Homey:
[image]

Here is Solinteg register table:
https://eshop.helion.cz/user/related_files/solinteg_modbus_protocol_mht-25-50-1.pdf

biemond, 2024-05-28:
indeed , they are on the same level as total pv.

[image]

biemond, 2024-05-28:
will change the decimal part tonight