biemond/solaredge.modbus #82: Huawei inverter

Knak79, 2024-06-04:
Hi,
I am able to connect to the inverter through Homey, but I keep getting the message "last value unknown" on every parameter. Any ideas on what have been setup wrong? I have used the IP of the dongle and port 30024 and ID 1.

biemond, 2024-06-04:
hey

I think the. ip should be local and we use 502 as port mostly .  modbus id 1 can be ok. 
Also make sure you don't connect too many apps to do modbus on the s dongle. dongle cannot handle a lot of load.

thanks

ruben9094, 2024-06-04:
Hi @biemond 
Sorry to bother you. I'm a nebie as well to this.
Same situation for me. All data points show up as last value unknown. This is the only modbus tcp device communication. Other devices communicate on modbus RTU directly to the inverter rather then the S-dongle.
Did a port scan, port 502 is open connection was succesful.
Found some other threads that the timeout should be significantly increased, even as far as 2000ms. Don't see this option.
Could you help? Having this data would be a game changer for me and would be greatly appreciated!

biemond, 2024-06-04:
hey, 

nice, just use port 502on that ip and I did all timeouts etc for you already

probably your modbus id = 1 for your inverter. can you try that

biemond, 2024-06-04:
s-dongle becomes an issue with 2 inverters on 1 s dongle. I don't support that because s dongle is unstable.

biemond, 2024-06-04:
can you send me a diagnostic report.

ruben9094, 2024-06-04:
@biemond 
Sure, should have thought about it to do it immediatly :)
8032C47B

biemond, 2024-06-04:
I think you send the homey 1 , you can go to my homey app and do it from there.

ruben9094, 2024-06-05:
@biemond my bad, 9d3b2c90-8a46-4274-83a7-939372e1a81d

biemond, 2024-06-05:
looks like connection is ok but sdongle does not connect to the modbus.
what is the modbus id you use and do you have another running program who read this data?

```
Note: this log has been manually submitted by a user.
    Log ID: 9d3b2c90-8a46-4274-83a7-939372e1a81d

    User Message:
    As discussed on github

    stdout:
    error with key: STORAGE_FORCIBLE_CHARGE_DISCHARGE_SOC
UserRequestError {
  err: 'Offline',
  message: 'no connection to modbus server',
  response: undefined
}
error with key: STORAGE_FORCIBLE_CHARGE_DISCHARGE_WRITE
UserRequestError {
  err: 'Offline',
  message: 'no connection to modbus server',
  response: undefined
}

```

ruben9094, 2024-06-05:
Dag Edwin

Dankje voor de troubleshooting.
Alles qua instellingen lijkt dus goed te zitten.
Modbus id =1

Ik heb een support case bij huawei geopend. Even afwachten wat zij wel/niet kunnen zien.
Alvast bedankt om menop het juiste spoor te zetten.

Grt
Ruben

Verstuurd vanaf mijn iPhone