biemond/solaredge.modbus #83: Geen data uit inverter Solar Edge

Jasper27NL, 2024-06-04:
Beste ontwikkelaar. 

Ik heb geprobeerd de info van de inverter naar homey te krijgen maar helaas wil het niet lukken. 

d2c7355f-d65a-4b1d-8c8b-f5918dd2cb68

Kun je zien wat er mis gaat?

biemond, 2024-06-04:
yes timeouts.  is de ip correct, is de port wel open en heb je een een firewall.   

je kan een port scanner gebruiken. verwijder de device eerst.  poort 1502 moet open zijn.  Je hebt toch wel modbus aangezet?

```
Error: connect ETIMEDOUT [192.168.86.52:1502](http://192.168.86.52:1502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16) {
  errno: -110,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '192.168.86.52',
  port: 1502
}
```

Jasper27NL, 2024-06-04:
Bedankt voor je snelle reply. 

Ik heb via een port scanner gekeken en poort 1502 geeft die niet aan als een open poort. 
Wel gaf die aan dat een andere poort open stond. 
Die heb ik nu gewisseld in de instellingen in Homey en nu werkt het. 
Of mag het niet via een andere poort dan de 1502?
Ik kon zo snel in de inverter niet vinden waar ik de poort kan wijzigen/openzetten. 
Modbus had ik wel aangezet

biemond, 2024-06-04:
ja het is 1502 of 502 meestal. of je hebt een andere poort op de inverter aangegeven