biemond/solaredge.modbus #84: Huawei connectie verbroken.

RayverPro, 2024-06-05:
Sinds vandaag is de Huawei connectie verbroken. Het lijkt na de laatste update ineens kapot te zijn.
De connectie was via poort 6607, nu lees ik dat daarvoor poort 502 gebruikt moet worden. Ook deze werkt niet bij mij.
Ook heb ik de software gedowngrade naar versie 3.0.115 maar ook dat heeft niet geholpen.

biemond, 2024-06-05:
hey,

vreemd , ik heb zelf geen huawei of s dongle.   maar s dongle is beroemd om connectie problem.  

stuur me maar een paar rapporten , ik zal ze doorsturen.

biemond, 2024-06-05:
maar zorg ervoor dat je niet je s-dongle overbelast of overvraagt.

RayverPro, 2024-06-05:
a190bb92-4bbb-4cc0-80fd-270271d20943

biemond, 2024-06-05:
is 6607 niet de web api poort, het moet de modbus poort zijn,  denk 502. 
```
[Driver:huawei] [Device:7bae2111-81f4-4847-91e7-1c600604a8d3] 192.168.178.245
Error: connect EHOSTUNREACH [192.168.178.245:6607](http://192.168.178.245:6607/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.178.245',
  port: 6607
}
```

RayverPro, 2024-07-04:
Na veel uitzoeken en proberen is vandaag het probleem opgelost. 

In de Inverter waren instellingen plots gewijzigd. (waarom is mij nog steeds niet duidelijk)
Ook heb ik een andere router aangeschaft, dit omdat de Ziggo Modem tegen zijn limiet aan liep qua aangesloten devices. 
Nu een Router (TP-Link) die goed samen gaat met Home Domotica & Homey. 

Wat ik erg vreemd vind maar wel de moeite waard vind nog even te vermelden, is dat mijn Huawei-SUN-2000 blijkbaar wel werkt op poort 6607 en helemaal niet op poort 502. Misschien is er iemand die mij dat uit zou kunnen leggen. Tevens een tip voor degene die ook problemen hebben deze Huawei Inverter werkend te krijgen.
[image]

biemond, 2024-07-04:
wat ik weet dat s-dongle kan de ethernet overnemen of de inverter kan het doen.   volgens mij als je meerdere inverters hebt is het handig om de sdongle modbus laten afhandelen, zodat je beide kan uitlezen. 

see this guide https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/667250677153415168-667213868771979264
and this https://solar.huawei.com/download?p=%2F-%2Fmedia%2FSolarV4%2Fsolar-version2%2Fcommon%2Fprofessionals%2Fall-products%2Fproduct%2FSmartDongle-WLAN-FE%2Fsupport%2Fpdf%2FSmartDongle-WLAN-FE-Quick-Guide.pdf