biemond/solaredge.modbus #85: Homey Pro app werkt niet

jbdegraaf, 2024-06-14:
Heb sinds kort een Homey Bridge Pro maar deze krijgt geen data binnen van mijn Solaredge inverter. De officiële app is geïnstalleerd maar er komt geen data binnen ook niet naar verloop van tijd of als de power optimizer actief zijn.

biemond, 2024-06-14:
hey, 

homey bridge doet geen wifi of lan (alleen zigbee, z-wave, BLE).  Je hebt een homey pro nodig

jbdegraaf, 2024-06-14:
Ik heb een Homey (Bridge) Pro.

biemond, 2024-06-14:
mooi , heb je modbus aangezet? en kun je me een diagnostic report sturen

biemond, 2024-06-14:
ja zie post 1 https://community.homey.app/t/app-pro-solaredge-growatt-tcp-modbus/45487 
via het inverter menu or in mijn geval . moet ik de lokale wifi aanzetten en dan via de app configuren.

jbdegraaf, 2024-06-14:
Ik kan die Modbus setting in het geheel niet terug vinden terwijl ik op de webserver zat. Ik zie hem niet Communications

Dit is de firmware informatie van mijn SolarEdge Inverter:

[image]