biemond/solaredge.modbus #86: Huawei dongle - Can’t connect

weide43, 2024-06-16:
Hi,

last week they installed solar panels and the Huawei battery with also a Huawei inverter.
I installed the app and searched for the name of the dongle on the router. I filled it in but nothing happened.

Report code: adb6eb34-fa0a-4e86-830e-6326067b3de8

(Nederlands kan ook)

Met Vriendelijke Groeten 
Danny

biemond, 2024-06-16:
Hey,

is poort 502 open op dit IP adres?  gebruik je een s-dongle?

Gr Edwin

```
[Driver:huaweiwithbatt] [Device:fa972dce-0b43-4e59-acbb-7c2b3be6ff68] 192.168.2.170
Error: connect ECONNREFUSED [192.168.2.170:502](http://192.168.2.170:502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '192.168.2.170',
  port: 502
}
```

weide43, 2024-06-16:
Hoe zet je poort 502 open?

biemond, 2024-06-16:
ik heb er zelf geen 1 maar kun je dit eens checken

https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/667250677153415168-667213868771979264

weide43, 2024-06-16:
> Hey,
>
> is poort 502 open op dit IP adres? gebruik je een s-dongle?
>
> Gr Edwin
>
> ```
> [Driver:huaweiwithbatt] [Device:fa972dce-0b43-4e59-acbb-7c2b3be6ff68] 192.168.2.170
> Error: connect ECONNREFUSED [...]

Ik gebruik een S-dongle

weide43, 2024-06-17:
Opgelost!
Dongle was verbonden maar de omvormer niet.

biemond, 2024-06-17:
nice