biemond/solaredge.modbus #88: Wrong data Huawei

joris666, 2024-06-30:
Dear,

I have a Huawei inverter + battery.
The SolarEdge app works perfectly and I receive all the data, which is approximately the same as the data from the FusionSolar app.
But as soon as the battery is 100% charged, the “active power” changes to a much larger value than in reality.
As soon as the battery drops back to, for example, 95%, it returns to the actual value comparable to that from the FusionSolar app.
Is this a bug?

Best regards,

biemond, 2024-06-30:
Hey, 

it can be an inverter bug or feature which they correct on the app. I just pass on what I read and give it to you,
I got the same on solaredge , when I restrict batt charging or batt is full , my houseload is doubled , somehow.

joris666, 2024-06-30:
If I understand correctly, so it has nothing to do with the SolarEdge app? But is it related to Huawei?

biemond, 2024-06-30:
yep, I just pass it on with no change