biemond/solaredge.modbus #89: Huawei Inverter + Battery. Flow cards "Then" not available?

AntondK88, 2024-07-02:
Hello,

The "when" (read) flow cards work perfectly for my Huawei Sun2000/Luna.
I'm just wondering why there are no "then" (write) flow cards available for the Huawei devices.
Is this not possible for the Huawei devices?
Or is this still a to-do item? If so, when could this be expected?

biemond, 2024-07-02:
Hey,

it is about #76  they have implemented some kind of authentication which is not described anywhere.  Huawei is the only 1 who is doing this.

thanks

AntondK88, 2024-09-21:
Hello Edwin,

I am able to write to my huawei sun200 through modbus (homey).
Previously I used the build in wifi connection and indeed, this is a read only connection.
Since yesterday I'm connected through a seperatly supplied connection dongle.
With this dongle I'm able to write to the Registers.

biemond, 2024-09-21:
oh really , I can add it again then. nice

Tuga67, 2024-11-29:
Hi @biemond 
First of all thanks a lot for the creation of this App. Super helpful!
Any idea when we could see these additional flow cards for Huawei?
Want to take lower Tibber prices to force charge the batteries now that winter is not so helpful with solar...

Thanks and keep the great work!

biemond, 2024-11-30:
let me add it again, you have a Lan connection or not?

biemond, 2024-11-30:
@AntondK88 @Tuga67 can you verify the device actions ?  if that works I will also add the flow actions.

Tuga67, 2024-11-30:
> let me add it again, you have a Lan connection or not?

Mine is conneceted with LAN

biemond, 2024-11-30:
nice can try the device actions.

Tuga67, 2024-11-30:
> @AntondK88 @Tuga67 can you verify the device actions ? if that works I will also add the flow actions.

Let me know when they are available to use again and I will test them. Thanks for your support!

biemond, 2024-11-30:
not triggers but on the device , 1st tab when you open the device.  lets see that 1st how it goes.

biemond, 2024-11-30:
I see they are not settable , let me fix that

biemond, 2024-11-30:
here is an new version https://homey.app/a/solaredge.modbus/test/

Tuga67, 2024-11-30:
Tried to install this one and still do not see the new cards.
Do I have to uninstall the previous version and install this one?

biemond, 2024-11-30:
No but you have to wait a bit until that version shows up. You have the Huawei with battery device or not

Tuga67, 2024-11-30:
OK. I will wait.
My system is a 10kW Sun2000 and 10kW (two 5kW batteries) Luna2000

biemond, 2024-11-30:
should be there by now, can you check for the app version  v3.0.167 , else maybe add it temporary again to see it if the device actions are showing up on the device.

Tuga67, 2024-12-18:
Hi @biemond,
Apologies but had to travel and did not have time for proper testing until yesterday.
Version I see is v3.0.169
My feedback until now:
- I can force charge using the fusionsolar web and app
    - super cumbersome and not user friendly
    - No way to automate = all is manual
    - takes ages to actually see if something is happening after I provided the commands
- Tried doing using the commands over your Homey app
    - All commands are visible as drop downs
    - First drop down level drives change to second level
    - At second level I cannot see if command was actually transmitted
    - Force charge does not work if the inverted is "sleeping" in standby
    - Works in the morning when the inverter starts waking up (some voltage in the panels)  even if the PV is not yet working
- I cannot see any of the AND or THEN flow actions
- Couple of questions:
    - Any chance you could share a couple of instructions on the drop downs and what each one is actually driving?
    - Any idea why I cannot see the AND and THEN flow actions?
    - Any chance that you could send a "wake on LAN" or another message for the Inverter and the Batteries to wake up?
Again, thanks for your great work and support

biemond, 2024-12-18:
ok, good 

yep did not do flows because I did not know if it works, so only device actions but what you do is postive. 
and is normal that you need to do multiple actions to get a certain  result.  that is how modbus works , sometimes I combine multiple actions but I don't know how it should work.

but you need to get your own answers or give me some guidance,  I got solaredge and only made this because huwaei web/api was not working as expected.

So I follow the huawei modbus docs and some HA repos but that is it. 

for AND you can use conditional ones from homey then actions when it is available and make sense and which in  the modbus docs.

wake on lan or something else I need to know what to send .

biemond, 2024-12-18:
here is the doc https://support.huawei.com/enterprise/de/doc/EDOC1100387581?currentPartNo=k004&togo=content&section=k004

Tuga67, 2024-12-19:
Great, Thanks.
I will do some tests tomorrow during the day and record each step. I will also try to review (from the interface definitions doc) which are the numbers and addresses that I would guess are behind each one of my steps. I then add some of the variables that I submit (e.g. the charging power).
Finally then create a logical sequence which works manually using FusionSolar web or app.

biemond, 2024-12-19:
here is what I do https://github.com/biemond/solaredge.modbus/blob/main/drivers/huaweiwithbatt/device.ts#L113

CGG1983, 2024-12-19:
When adding unit as Huawei Inverter + Battery I do not get the options to "write", when I add unit as Huawei 2 inverter S-dongle + battery I get the "write" options.
Question:
- why is the options different for the two above (my understanding is 1 vs 2 inverters, but why S-dongle as difference)?
- when adding as the second option I get the following error when trying to force charge/discharge. Missing Capability Listener: storage_force_charge_discharge. Why do I get this?

biemond, 2024-12-19:
need to check , it looks ok to me but clearly not 

https://github.com/biemond/solaredge.modbus/blob/main/drivers/huaweiwithbatt/device.ts#L113

https://github.com/biemond/solaredge.modbus/blob/main/drivers/huaweidouble/driver.flow.compose.json#L127

biemond, 2024-12-25:
found it , was re-using huawei batt caps for s-dongle double. and enabled the device actions which caused this issue on the double device. 

so disabling them again and replace them with flow actions then.

biemond, 2024-12-25:
can you try this https://homey.app/a/solaredge.modbus/test/

CGG1983, 2024-12-25:
Hi,

Thansk alot!

I tested both the "Huawei Inverter + battery" and the "Huawei 2 Inverters S-dongle + battery".
- In both cases I get the "then" commands and can put up the flows in Homey
- Engaging the commends is not resulting in any error messages in either case
- However, it seems like when installed as "Huawei 2 Inverters S-dongle + battery" is the only option where it actually receives the commands and changes status (I tested forcible charge/discharge since this is the function I am searching for
- When installed as "Huawei Inverter + battery" the status does not seem to change when commands are sent

When I use the "Huawei 2 Inverters S-dongle + battery" I select ID1 for the one wiht battery and ID3 for the second one.

Info:
- I am only running 1 inverter with 10 kWh battery and Smart meter
- Inverter is connected by WIFI

For me the "Huawei 2 Inverters S-dongle + battery" seems to be the best option even when running one inverter.

Thanks again for setting this up!

biemond, 2024-12-25:
interesting basically it is the same.  but I guess 2 inverter s-dongle I am very slow with the retry like every 2,3 minutes  and do both instead of every 55 seconds. So I guess s-dongle cannot handle this frequent ask.   I can add this polling freq to huawei batt so you can also put it to 2 min.

but more interesting , how is it working on wifi. was thinking it is a lan option only because it will ask for a password when it is wifi.

CGG1983, 2024-12-25:
Can it be so that the WIFI option works because I have set Modbus-TCP to ”enabled unrestricted”?

biemond, 2024-12-25:
cool , awesome. never knew this option

wachaudhary, 2025-01-24:
I have been testing the experimental version of the Homey app (version 3.0.173) with my current setup, which includes a Huawei 12kW inverter and a 14kW Huawei Luna S1 battery. However, I encountered an issue where the "Force Charging ON" function does not trigger the system as expected. Despite using the available commands within the app, the action fails to initiate and eventually results in an error after some time.

@Tuga67 , I wanted to ask if you were able to successfully enable force charging on your system. Specifically, did the battery start charging as expected once this function was activated?

[image]

mackewigren, 2025-03-20:
@Tuga67 or @wachaudhary 
Hi, have you gotten this to work? I do not get anything to happen. I just wanted to try for example to force charge. Or to just change any setting. Nothing happens and no error messages. Can you share any simple example that works? Have not understand if I need to run several flow cards to trigger.

Over time I am mostly interested in change settings between:
1) Start charging
2) Start discharging but not force discharge
3) Force discharge

If I only have these commands my plan is to build some kind of smart flow that calculates when to change the settings.

mackewigren, 2025-03-20:
> I have been testing the experimental version of the Homey app (version 3.0.173) with my current setup, which includes a Huawei 12kW inverter and a 14kW Huawei Luna S1 battery. However, I encountered an issue where the "Force Charging ON" function does not trigger the system as expected. Despite using the available commands within the app, the action fails to initiate and eventually results in an error after some time.
> 
> [@Tuga67](https://github.com/Tuga67) , I wanted to ask if you were able to successfully enable force charging on your system. Specifically, did the battery start charging as expected once this function was activated?
> 
> <img alt="Image" width="826" src="https://private-user-images.githubusercontent.com/191789245/406604893-78afdd3d-a7a1-475d-aa89-8ec399a86b2c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI0NjI5MTYsIm5iZiI6MTc0MjQ2MjYxNiwicGF0aCI6Ii8xOTE3ODkyNDUvNDA2NjA0ODkzLTc4YWZkZDNkLWE3YTEtNDc1ZC1hYTg5LThlYzM5OWE4NmIyYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyMFQwOTIzMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zN2Q5MzM5ODU1YjA4NWFmMzhjZTQ1OTk0NDhlYWZmNTg4ZWFhZjBhYjlhYzViYjAxYWYyZTViMTBiZGZlMzkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Y6kiW6Bxr5qhdkXJAU4xz9e81U2y2HbZARfpayt8N0o">

Another question. What app are you using to publish your values as in your screenshot? Can you add your own images (like the apple logo)?

CGG1983, 2025-03-20:
> [@Tuga67](https://github.com/Tuga67) or [@wachaudhary](https://github.com/wachaudhary) Hi, have you gotten this to work? I do not get anything to happen. I just wanted to try for example to force [...]

Hi,
I had some challenges in getting a single command to reach . so i am sending repeat commands. This now works:

[image]

biemond, 2025-03-20:
if you use s-dong;e, it can explain this. cheap but so horrible for automation (slow and can only handle 1 connection at the time).

CGG1983, 2025-03-20:
Yes. I am using s-dongle and not wired connection.

CGG1983, 2025-03-20:
> Thanks [@CGG1983](https://github.com/CGG1983) I tried but cannot get ti to work. Is it Huawei you are using? (guess so)

Yes using huawei. 
10 kw inverter and 10kwh battery 
All equipment from late 2022
Running on homey 2019 and have fixed ip for my dongle/inverter

Tuga67, 2025-03-20:
Hi all, sorry for the absence. Back in the country now.
Just tried the @CGG1983 solution but it also does not work on my system.
I am currently using version 3.0.178 as per yesterday's update (thanks @biemond for your continued great work!).
My system is a SUN2000-10KTL-M1 running V100R001C00SPC168
Batteries are LUNA2000-5KW-C0 running V100R002C00SPC632
Dongle is SDongleA-05 running V200R022C10SPC200
This is the flow I created as a copy of your test:

[image]

Tuga67, 2025-03-20:
I am also using the dongle via WiFi (sometimes wired solution is not reliable).
However, I did not fixed the IP to any of the Inverter or Dongle.

Tuga67, 2025-03-20:
I wonder...
Could it be that the command has to, either have more parameters or, be part of a string of parameters?
Let me explain. When I Force Charge using FusionSolar on the web, it shows (as an example):
- Charge / Discharge: Charge
- Forced charge power (kW): 1.000
- Setting mode: Duration
- Forced charge/discharge period (mins): 60
After I press "Save" (and I do not get a timeout), I start to visualise additionally:
- Remaining charge/discharge time (mins):
- Charged energy (kWh):
So, could it be that we need additionally to provide Forced charge power (kW): 1.000, Setting mode: Duration and Forced charge/discharge period (mins): 60 so that it works?

I can confirm that using the FusionSolar screen that you see below, the batteries start to charge from AC without any issue:

[image]

Note:
Charge from AC on Parameters Settings has to be enabled for this to work (I have 4 panels on the garden with a microinverter and this is the only way the batteries charge from these panels as well)

biemond, 2025-03-20:
for me it is this crappy s-dongle while doing modbus,  after connect if that works , have to wait for 5 seconds , every call has a 5 seconds timeout. have to wait 250ms in between.  normal retrieve takes a minute, make sure nothing else is going on. 
but it is cheap at least compared to emma. but that takes under a second.

JorgenSeemann, 2025-06-16:
> I wonder... Could it be that the command has to, either have more parameters or, be part of a string of parameters? Let me explain. When I Force Charge using FusionSolar on the web, it shows (as an [...]

Same for me with my SUN2000-10KTL-M1+Battery.
I can change it via Fusionsolar (same way) but it's completly ignoring changes that I try to make via this app.
For me it makes no difference if it's connected via WIFI or LAN.

biemond, 2025-06-17:
https://github.com/biemond/solaredge.modbus/blob/main/drivers/huaweiwithbatt/driver.flow.compose.json

biemond, 2025-06-17:
on the display of the device you mean?