biemond/solaredge.modbus #90: Set Export Limit Max Value

jetspot, 2024-07-08:
Hello Biemond,

First of all, what beautiful software you have created!!

I, and probably many more people, are very happy with this.
Thank you !!

I have a small request.
can you set the maximum value of
Set Export Limit" to 50000 watts.
this is for solaredge with a battery.

My system is larger than 25,000 watts.
i have 1x a Solaredge 10 RWB and 2x Solaredge 10 RWS
30000 KW
And I can no longer get my inverters to maximum power.

Kind regards
Marcel Smit

[image]

jetspot, 2024-07-08:
> Hey,
>
> nice. ok. no problem.
> just 1 question why do you set to it 50k. I guess you can disable export
> limit instead?
>
> thanks

Yeah, that's what I thought too, but it doesn't work
when I look in the set app it is still at 25,000 watts
I have noticed that if I set the value higher than the value of the
inverter, it sets the value to the maximum system value in my case 30000
watts

thanks for your response

PS. Are you from the Netherlands

ja dat dacht ik ook maar dat werkt niet,
als ik in de setapp kijk staat hij nog steeds op 25000 watt
ik heb gemerkt dat als ik de waarde hoger zet dan de waarde van de
omvormer dan zet hij de waarde op de maximale systeem waarde in mijn geval
30000 watt

dankt voor uw reactie

-----------------------------------------------
Met Vriendelijk Groet,

M.Smit
Smit Techniek en Service
06-55341155

biemond, 2024-07-08:
ok, 

dus als je limit mode op disabled zet. dan is het nog steeds 25k?  volgens mij gaat hij dan naar de default of hij negeert deze waarde.
[image]

je kan ook proberen hem op max self consumption te zetten.

jetspot, 2024-07-10:
Ik ga het even uitproberen Bedankt Verstuurd vanaf mijn iPhone

Screenshot.2024-07-08.at.17.24.45.png (view on web)

jetspot, 2024-07-10:
Ik heb het even gestest maar dit werkt niet als ik dit in een flow van
homey doe
wel als ik het aanklik net zo als op jouw voorbeeld

-----------------------------------------------
Met Vriendelijk Groet,

M.Smit
Smit Techniek en Service
06-55341155

biemond, 2024-07-10:
als je  limit mode op disabled zet is hetzelfde als de menu

[image]

jetspot, 2024-07-11:
goedemorgen

nee werkt bij mij niet

-----------------------------------------------
Met Vriendelijk Groet,

M.Smit
Smit Techniek en Service
06-55341155

biemond, 2024-07-11:
als je meerdere flow acties doet kun je voor de zekerheid een 10 sec pauze tussen de acties. 

en daarna mij een diagnostic report sturen.

Gr

jetspot, 2024-07-12:
Hallo

hier de diagnose code

ebf9c926-9ff4-4505-bc85-6dfe4524578d

-----------------------------------------------
Met Vriendelijk Groet,

M.Smit
Smit Techniek en Service
06-55341155

biemond, 2024-07-12:
ziet er goed uit, het is disabled dus die 25k limit moet niet uitmaken. 
```
storage_control_mode 1 xxx Storage Control Mode
storage_accharge_policy 1 xxx Storage AC Charge Policy
storage_accharge_Limit 0 xxx Storage AC Charge Limit
remote_control_command_mode -1 xxx Remote Control Command Mode
remote_control_charge_limit 5000 xxx Remote Control Charge Limit
remote_control_command_discharge_limit 5000 xxx Remote Control Command Discharge Limit
remote_control_command_timeout 3600 xxx Remote Control Command Timeout
remote_control_default_command_mode 0 xxx Storage Charge/Discharge Default Mode
active_power_limit 100 xxx Active Power Limit
advancedpwrcontrolen 1 xxx Advanced Power Control En
reactivepwrconfig 4 xxx Reactive Power Config
export_control_mode 0 xxx Export control Mode
export_control_limit_mode 0 xxx Export control limit Mode
export_control_site 25000 xxx Export control site limit
powerreduce 100 xxx Power Reduce

```

ik zie een paar  unhandledRejection

is de device van de app hetzelfde als waar je flow triggers op voert?

jetspot, 2024-07-13:
Het vervelende is alleen Als ik in de omvormer kijk via de SetApp Dan is er niks veranderd in de settings Dan staat hij nog steeds op de limit van bv 100 watt

jetspot, 2024-07-13:
Ik zou het toch heel fijn vinden als de waarde wordt verhoogd naar bv 50000 watt

biemond, 2024-07-14:
kun je deze eens testen https://homey.app/a/solaredge.modbus/test/

jetspot, 2024-07-15:
Nee het werkt niet

Ik moet echt de waarde hoger zetten dan de site limiet 

[image]

Hij blijft gewoon op 100 Watt staan

jetspot, 2024-07-15:
Ik heb nu dit er in staan en dat werkt prima

[image]