biemond/solaredge.modbus #91: Whats the meaning of "Batterie wird mehr als %" ?

JanSchaefer79, 2024-08-09:
Hello, i have a Sungrow Hybrid with Battery.
What is the meaning of function 
If  "Batterie wird mehr als %" 
If "Batterie wird weniger als %" ?

Is it the stored kwh in percent or is it the Battery Quality in percent ?

Is there somewhere discription for the German Cards ?

with best Regards

biemond, 2024-08-09:
Hey , 

should be SOC flow trigger and not SOH. I think these are auto generated by homey , not by me and I guess it uses battery %

JanSchaefer79, 2024-08-09:
Ok, i will try later again. Thanks for your comment. The translation is sometimes little bit confusing with the cards

biemond, 2024-08-09:
yes , these are auto generated but most of my flow triggers are translated by your countryman.