biemond/solaredge.modbus #93: Sungrow don't report anymore

hmarkstrom, 2024-08-31:
Hi,

Sungrow has stop reporting. The last time was 23 days ago.

biemond, 2024-08-31:
hey, can you send me a diagnostic report. 

maybe your ip has changed or connected another app to your inverter

hmarkstrom, 2024-09-11:
I have checked the WiFi on my Sungrow hybridconverter and notihng was changed.

biemond, 2024-09-12:
ok send me a report and a do port scan at least. because something has changed on your side.

hmarkstrom, 2024-09-19:
I have reported to Sungrow of my problem and yesterday they install new firmware for all devices in my hybridcontroller and I delete the app and installed it again and still no values is reported.

biemond, 2024-09-19:
can you send me an diagnostic report, like to see if modbus port is open/enabled  and ip is correct. it has to be something like that.  you can also do a portscan.

hmarkstrom, 2024-09-21:
I have found the problem. Sungrow updated the WiFi dongel for a month ago and after that it was not connected to WiFi anymore.
Have reported the problem to Sungrow.