biemond/solaredge.modbus #94: App werkt niet op een oude homey.

Bigboedha, 2024-08-31:
Hoi,
Ondanks dat er staat dat de app ook werkt op een early Homey van 2016 werkt dit niet. Nu wil ik eventueel wel een nieuwe homey kopen. Ik heb alleen wel een paar vragen over de app.

Ik heb een Solaredge Storedge met een 13,8 kw accupack. De inlinemeter van Solaredge, die draadloos is aangesloten. 
Gaat het dan werken? Of moet de inlinemeter via modbus en dan bekabeld?

Verder is het mogelijk om met de app de ontlaad snelheid van de accu's aan te passen? Nu wordt er steeds ontladen met 5 kw per uur.

gr
Alex

biemond, 2024-09-01:
hey,

kun je mij een report sturen? eens kijken wat het is. krijg je nog wel homey updates binnen en op welke versie draai je nu.

draadloos is lastig met timeouts , modbus en een kabel op de meter werkt beter.  het is niet al te moelijk.

ja met remote control kan je van alles regelen zoals charge / discharge van je batterij. 
export / import restricties , laden vanuit grid.
[image]

Bigboedha, 2024-09-01:
Homey Versie 0.0.0-update.inc zeg de App.
App versie is 8.4.0.1538
Node.js-versie v12.22.11

Als ik probeer de Soloredge app te installeren geeft de Homey app alleen op Homey Pro.

Wat voor een report zou je willen hebben? Ik zie alleen een code van een diagnostisch rapport B30664CF.

Heb de homey weer uit de motteballen gehaald. Deze heeft al een tijdje niks gedaan. Eigenlijk alleen omdat ik je app zag voor de solaredge.

biemond, 2024-09-01:
ok , ja het is geen pro.   weet niet waar je woont maar je kan mijn homey pro 2019 overnemen voor 50 euro. 
Je kan ook Home assistance ergens installeren en met hun flows (ietjes moeilijker , maar wel gratis) 

ook wel handig als je advanced flows bij koopt, die oude manier van flows is lastig.

biemond, 2024-09-04:
ja met een nieuwe krijg je advanced flows en doe 3 draadjes van een ethernet kabel voor de meters naar inverter aansluiting.

Bigboedha, 2024-11-18:
Hoi,
Het project is gestart, de Homey is binnnen en de 3 draadjes zijn van de
Solaredge Inline Meter naar de omvormer getrokken.

Alleen ik begrijp de flows niet helemaal. Ik zit bij Tibber als mijn
stroomleverancier. Dit heeft ook een app in de Homey omgeving. Nu wil ik
graag dat mijn accu's gaan laden als de stroom bij Tibber goedkoop is en
ontladen als de stroom duur is of als er vraag is in huis.

Heb je misschien een voorbeeld van een flow structuur hoe ik dit moet
opzetten?+

Moet je de flow's trouwens apart starten of gaat dit automatisch?

Alvast bedankt,

mvg
Alex Ankersmit

biemond, 2024-11-18:
sorry everyone is doing this differently. 

some do hour by the power for the prices and after that do some flow actions.  
if you do anything automatically like checking some power etc and then do something, make sure you use average values.  else maybe you trigger too many actions on your inverter.

basically solaredge has 2 actions , general 1 like disable , max self consumption or remote
if you want to control charging , you need to put it in remote and than select charge from grid.   besides this you can also the charging or discharging .  Every action is max for 6 hours and then it goes back to the default. 

so with advanced flows try some actions to see how it works .

biemond, 2024-11-18:
```
      if (type == 'storagecontrolmode') {
        // 0 – Disabled
        // 1 – Maximize Self Consumption – requires a SolarEdge Electricity meter on the grid or load connection point
        // 2 – Time of Use (Profile programming) – requires a SolarEdge Electricity meter on the grid or load connection point 3 – Backup Only (applicable only for systems support backup functionality)
        // 4 – Remote Control – the battery charge/discharge state is controlled by an external controller
        const storagecontrolmodeRes = await client.writeSingleRegister(0xe004, Number(value));
        console.log('controlmodewrite', storagecontrolmodeRes)
      }

      if ( type== 'storageacchargepolicy') {
        // 0 – Disable
        // 1 – Always allowed – needed for AC coupling operation. Allows unlimited charging from the AC. When used with Maximize self-consumption, only excess power is used for charging (charging from the grid is not allowed).
        // 2 – Fixed Energy Limit – allows AC charging with a fixed yearly (Jan 1 to Dec 31) limit (needed for meeting ITC regulation in the US)
        // 3 – Percent of Production - allows AC charging with a % of system production year to date limit (needed for meeting ITC regulation in the US)
        const storageacchargepolicyRes = await client.writeSingleRegister(0xe005, Number(value));
        console.log('storageacchargepolicy', storageacchargepolicyRes)
      }

      if (type == 'storagedefaultmode') {
        const storagedefaultmodeRes = await client.writeSingleRegister(0xe004, Number(4));
        console.log('controlmodewrite', storagedefaultmodeRes)

        // set timeout to 6 hours, done in seconds
        const remoteTimeout = await client.writeMultipleRegisters(0xe00b, [ Number(21600), 0 ]);
        console.log('remote_control_command_timeout', remoteTimeout);
        // 0 – Off
        // 1 – Charge excess PV power only.
        // Only PV excess power not going to AC is used for charging the battery. Inverter NominalActivePowerLimit (or the inverter rated power whichever is lower) sets how much power the inverter is producing to the AC. In this mode, the battery cannot be discharged. If the PV power is lower than NominalActivePowerLimit the AC production will be equal to the PV power.
        // 2 – Charge from PV first, before producing power to the AC.
        // The Battery charge has higher priority than AC production. First charge the battery then produce AC.
        // If StorageRemoteCtrl_ChargeLimit is lower than PV excess power goes to AC according to NominalActivePowerLimit. If NominalActivePowerLimit is reached and battery StorageRemoteCtrl_ChargeLimit is reached, PV power is curtailed.
        // 3 – Charge from PV+AC according to the max battery power.
        // Charge from both PV and AC with priority on PV power.
        // If PV production is lower than StorageRemoteCtrl_ChargeLimit, the battery will be charged from AC up to NominalActivePow-erLimit. In this case AC power = StorageRemoteCtrl_ChargeLimit- PVpower.
        // If PV power is larger than StorageRemoteCtrl_ChargeLimit the excess PV power will be directed to the AC up to the Nominal-ActivePowerLimit beyond which the PV is curtailed.
        // 4 – Maximize export – discharge battery to meet max inverter AC limit.
        // AC power is maintained to NominalActivePowerLimit, using PV power and/or battery power. If the PV power is not sufficient, battery power is used to complement AC power up to StorageRemoteCtrl_DishargeLimit. In this mode, charging excess power will occur if there is more PV than the AC limit.
        // 5 – Discharge to meet loads consumption. Discharging to the grid is not allowed. 
        // 7 – Maximize self-consumption
        const remotecontrolwrite = await client.writeSingleRegister(0xe00d, Number(value));
        console.log('remotecontrolwrite', remotecontrolwrite);
      }

```

Bigboedha, 2024-11-18:
Nu ben ik echt een leek op dit gebied. Ik heb de bovenstaande tekst
gecopieerd en in de Homey gezet als script.
Zoals ik het zie, moet ik waar value staat een getal in vullen wat vermeld
staat erboven achter de //.

biemond, 2024-11-18:
nee , gebruik als flow actions on je solaredge device 

[image]