biemond/solaredge.modbus #95: Growatt stops reporting

ostracus, 2024-09-08:
Hi 

I am really fund the app, and hence its even worse, that it stopped working :-( My Growatt Inverter stopped reporting 10 days ago, don’t know why. Checked all parameters, including IP and Modbus config, but I don’t see or recall that anything was changed. An update to Homey or? Still have access to the inverter through Shinephone, but just lost the modbus connection. Any suggestions?
BR M

biemond, 2024-09-08:
Hey, did you send me this diagnostic report? 

is your ip changed with dhcp?  did you do a port scan on ip and port?

```
[Driver:growatt] [Device:7825b3d2-fa5b-4cf5-abf2-7ea6acf2fbc7] 192.168.10.18
Error: connect ETIMEDOUT [192.168.10.18:502](http://192.168.10.18:502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16) {
  errno: -110,
  code: 'ETIMEDOUT',
  syscall: 'connect',
  address: '192.168.10.18',
  port: 502
}
```

biemond, 2024-09-09:
ok, please make sure you don't have any other app reading the same data and use a port scanner to see if the port is open