biemond/solaredge.modbus #100: Do not get any readings in Homey

Swamulf, 2024-09-20:
Hi, I am new to this app and just enabled Modbus on my Huawei inverter. After installing the app i do not get any readings. What might be the issue?

I am using a wifi extender which the inverter dongle is connected to (it shows up with a different name). The homey is connected to the main wifi (which is also connected to the extender). Can this be an issue?

[image]

biemond, 2024-09-20:
Hey, can you send me a diagnostic report? I think homey cannot reach it or maybe some timeout issue.

You can also try a port scanner

Swamulf, 2024-09-20:
Thank you for your answer!
9e57cc1f-8cbe-46d1-93ba-56025fc6e95b

What is a port scanner?

biemond, 2024-09-20:
hey, 

looks like , s-dongle cannot connect or so . I think we need to change something on inverter or s-dongle.

but ip and port is ok I guess.

```
[Driver:huaweiwithbatt] [Device:f21e27b7-a850-44e9-a7e1-c5fb21499ae0] 192.168.68.107
Connected ...
{
  host: '192.168.68.107',
  port: 502,
  unitId: 1,
  timeout: 115,
  autoReconnect: false,
  logLabel: 'huawei Inverter Battery',
  logLevel: 'error',
  logEnabled: true
}
Client closed
error with key: inputPower
UserRequestError {
  err: 'Offline',
  message: 'no connection to modbus server',
  response: undefined
}
```

Swamulf, 2024-09-24:
Thank you! Do you think there is a setting i should change?

biemond, 2024-09-24:
you need to check your s-dongle and inverter settings.   or maybe reboot of those can help.

Swamulf, 2024-09-24:
The modbus setting on the dongle is enabled so i'm not sure what other settings there are. 
What should the device ID be set to when added to Homey? Port is 502.
Can it be an issue that I use a wifi extender for the inverter and that Homey is connected to the main wifi?

biemond, 2024-09-24:
probably,

I saw on others it can be something on the s-dongle like gateway/router address issues.  it had to be something with networking on s-dongle