biemond/solaredge.modbus #102: Smart meter values.

szferi91, 2024-10-03:
Hello, 

I have 2 Huawei inverters, I have the information for both of them. My question would be if it's possible to read out the data from the smart meter linked to the inverted as that would give me the information about the Atvive Power. If that's negative I'm pulling from the grid, if positive, I'm feeding into the grid.

Best Regards

biemond, 2024-10-03:
Hey,

inverter don't have an api or or where you can set those values plus maybe risky when it hangs.  You need add some modbus meters and connect them to your inverters with rs482.

or maybe sdongle or emma can do this for you?

szferi91, 2024-10-03:
It is connected to the inverter with rs482 and has the sdongle as well installed. My question is if there is any possibility of having the information in the homey app

biemond, 2024-10-03:
it is there on (house) load etc and needed when you have a battery or do export restrictions.  But I won't add any external p1 meter.