biemond/solaredge.modbus #103: cannot add AND card in HomeyPro flow (WHEN ... AND ....)

cristeam, 2024-10-09:
Hello,

I want to create a flow like WHEN "Battery becomes greater than 80%"  AND "Battery Charge becomes greater than 3kW" THEN ... something.
Unfortunately, I can see your app "solarEdge + Growatt TCP modbus" only for WHEN card. I cannot see App in AND card!? 
[image]
[image]
[image]

biemond, 2024-10-09:
hey, 

I think you can also use a logical variables .  also I think advanced flows it support this. 
also what is your type of inverter

thanks

cristeam, 2024-10-09:
inverter is SUN2000-6KTL.
yes, I can use advanced flow, but then it's not that easy to change thresholds for cards ... I hope that's a simple fix for you ... other apps such as shelly = allow in both WHEN and AND sections ...
I am eager to help in testing/debugging if you want to fix it ...

voroniys, 2025-02-20:
I'm missing this functionality too. Yes it is possible to invent some workarounds for some cases, but it would be incomparable much easier if this app will appear in AND part of workflows as well.
And of course I can help with testing as well if you decide to fix this issue.

biemond, 2025-02-20:
it is too much work and to maintain it when homey already provides.