biemond/solaredge.modbus #104: freezing values from multiple devices of the App in HomeyPro (cascaded SUN2000 inverters)

cristeam, 2024-10-09:
Hello,

I have 2 SUN2000-6KTL inverters connected in cascaded setup: Master that has smartpowermeter + battery has ID=3, and a slave inverter has slaveID=1. The external connection is a Wired dongle with 1 IP in LAN (in inverter-master). 
When using a modbus simulator, I can access registers from each inverter by using the same IP address and only have 2 different slaveID (1 or 3).
In HomeyPro: I added 1 device with IPx, ID =3  type sun2000 & batt => and I get the correct data.
When I add another device with the same IPx, ID=1, type sun2000 => I see values only when HomeyPro reboots. Then I sometimes see only the active power updating, but battery-level => it never gets updated. Only if I reboot homey.

cristeam, 2024-10-09:
I can do any testing you need.
I suggest we need for these slow sdongle to:
- have a large modbus-timeout value (2 seconds) - better would be to have this parameter in advanced section for device, next to ID;
- to change the polling interval (or provide this parameter to App.
- Another issue would be to chain the modbus queries with a certain delay (1sec) between queries.

biemond, 2024-10-09:
yep, already have a slow startup like 5 sec and 2 sec on reading any register.   poll once per minute. 

indeed need to share the connection and do 1 by 1 .  
I need some test environment for that .