biemond/solaredge.modbus #105: Abnormaal communicatie tussen beheersysteem en apparatuur

eeckelaertyannick, 2024-10-26:
Alles met Homey lijkt te werken maar ik kan geen verbinding meer maken met fusion solar. Seetds foutmelding 999999909 abnormale Abnormaal communicatie tussen beheersysteem en apparatuur. Ik geraak er niet van af iemand idee? Ik heb de stick er in zitten voorheen geen last. Misschien een instelling?  Alvast dank.

biemond, 2024-10-27:
hey , is dat huawei with sdongle ? I heb geen idee, heb zelf solaredge.

maar heb je een firewall?  anders misschien homey even uitzetten of een factory reset

eeckelaertyannick, 2024-10-27:
Het is inderdaad een Huawei S-Dongle. Deze was er niet maar heb deze zelf geïnstalleerd. Ik heb al een factory rest gedaan. Tot nu toe geen nut. Geen firewall draaien enkel de modem zelf heeft iets basis. Maar het was niet voor ik de dongle geïnstalleerd had. Heb de vraag ook al even bij de installateur geplaatst. Met vriendelijke groet,Yannick Eeckelaert

closed 2025-05-30 by eeckelaertyannick