biemond/solaredge.modbus #107: Modbus production data sometimes decreases / becomes lower

Haandrwalt, 2024-11-13:
Last weeks, I experienced 3 times that the production values suddenly dropped. Sometimes a small amount during the day and all the time a bigger drop long after production stopped. These modbus register values should of course only go up and not down. I did experience some WiFi connection issues during the day, but do not see how that can be related (assuming the wifi communication does not have any influence on the converter processing and modbus updating). It looks like something is wrong with the converter, but it could also be an unknown write action from the Homey app.. (!?). See attached example graph.
[image]

Haandrwalt, 2024-11-13:
[image]
Similar behavior on nov 3rd

Haandrwalt, 2024-11-13:
No, not a lot of changes to the inverter. Once or twice a year a firmware upgrade, but I leave it alone as much as possible.
There seems to be a relation between loss of network connection during the day (WiFi) but I don’t see a link to register values being incorrectly updated later that day. Have had 3 similar experiences recently.
I assume there are no modbus write actions from the Homey app?

biemond, 2024-11-13:
cool.  then I think they store some data on solaredge and during the black out your inverters works as expected but when back online , it looks like solaredge data is preferred and local is ignored.

Haandrwalt, 2024-11-13:
SolarEdge support confirmed they do NOT update the converter data. Also the black out was for a short period of time. I do not see a link between WiFi outage and wrong register overwrites but will continue to investigate. Issue is also open as a support case with SolarEdge.

biemond, 2024-11-13:
interesting.  but then inverter did not update these registers because of wifi or some reboot and current values are not being stored.. 
total is just a register I read,  total day is calculated by me.

Haandrwalt, 2024-11-15:
I keep monitoring more intensive. Today I noticed small dips in the Total Day Yield graphs.
[image]

biemond, 2024-11-15:
interesting, looks like some reboot , goes to previous value, recovers from this issue and restores the values.

for me at 12:01 midnight on homey  I get the total value and store it . use that value for total daily.  so it is not homey at least.

Haandrwalt, 2024-11-15:
SolarEge app reports 1,12 kWh on this cloudy day, Homey report 1,11 kWh. Normally this is perfectly in sync, so looks like the dips did impact the daily production registration on Homey(?). Nothing special happened around 14:15. No updates, no reboots, no network connection issues. Strange …