biemond/solaredge.modbus #108: Sungrow string inverter

erikfriman, 2024-11-17:
First of all, thanks for a great app! I’m using it to connect my Sungrow hybrid inverter(SH8) + battery but also my “regular” Sungrow string inverter (SG8). Though my non-hybrid inverter is not officially supported it still reports “solar power” and “total yield”, but nothing else. My ask is to add support for regular Sungrow string inverters as well. Thanks, and keep up the great work!

biemond, 2024-11-17:
hey ,

interesting. can you send me a diagnostic report also do you have a doc about your modbus registers.

thankd Edwin

biemond, 2024-11-18:
nice,  let me check SH8 and see which version I developed this against.

biemond, 2024-11-18:
I think I need for 1 of your inverter a plain version with no batt or meters.   and for the other we need to check, almost there. 

can you remove your normal device so only the other is there.  maybe restart the app and send me a report after x minutes.  lets fix that 1 1st.

biemond, 2024-11-18:
can you do the other way around and see if with batt is ok.  after that will strip it and make a normal sungrow device.

biemond, 2024-11-18:
checking.

biemond, 2024-11-18:
I think I have to remove these , don't know what will be left here 

```
error with key: battery_power
error with key: loadpower
error with key: exportpower
error with key: outputPower
error with key: active_power_limit
error with key: pvTodayEnergy
error with key: pvTotalEnergy
error with key: temperature
error with key: battsoc
error with key: bmshealth
error with key: batttemperature
error with key: today_battery_output_energy
error with key: total_battery_output_energy
error with key: today_battery_input_energy
error with key: total_battery_input_energy
error with key: today_grid_import
error with key: total_grid_import
error with key: today_grid_export
error with key: total_grid_export
error with key: Runningstate
checkRegister result
error with key: emsmodeselection
error with key: charge_discharge_command
error with key: charge_discharge_power
error with key: max_soc 
error with key: min_soc
error with key: export_power
error with key: export_power_enabled

```

biemond, 2024-11-18:
let me check the doc for SG8

biemond, 2024-11-18:
what do you like to see from this page , the 5000 registers https://github.com/mvandersteen/SungrowInverter/blob/794cb9e52fb07b062c75e58112823d6d1a211c75/sungrowinverter/configs/string.py#L134

erikfriman, 2024-11-18:
A bit of guesswork on some of the below as I’m not sure what they actually do. I guess 5017 is the same as solar power in the app!?

What i originally was missing was actually just 5003 but I couldn’t resist adding more when I had the chance…

ModBusRegister(5003, "daily_energy_yield", "U16", 0.1, KILO_WATT_HOUR), 
ModBusRegister(5093, "daily_export_from_pv", "U32", 0.1, KILO_WATT_HOUR, valid_inverters=[0x013C,0x013E,0x013F,0x0142,0x0143,0x0147,0x0148,0x0149,0x2C0F]),
ModBusRegister(5017, "total_dc_power", "U32", unit_of_measure=WATT, description="PV power that is usable (inverter after inefficiency)"),
ModBusRegister(5101,"daily_direct_energy_consumption", "U32", 0.1, KILO_WATT_HOUR, valid_inverters=[0x013C,0x013E,0x013F,0x0142,0x0143,0x0147,0x0148,0x0149,0x2C0F]),
ModBusRegister(5128, "monthly_energy_yield", "U32", 0.1, KILO_WATT_HOUR),

erikfriman, 2024-11-23:
Anything else I can do to assist? 😊

biemond, 2024-11-23:
No, I am fine only  need to find some time probably coming week

biemond, 2024-11-26:
let me make a test version.

biemond, 2024-11-26:
can you try this 1 https://homey.app/a/solaredge.modbus/test/
and send me a report after a few minutes.

erikfriman, 2024-11-26:
[image]

erikfriman, 2024-11-26:
Looks promising 😊

biemond, 2024-11-26:
Cool, why are you limiting yourself to 80%

biemond, 2024-11-26:
Can you send me a report?

erikfriman, 2024-11-26:
Report sent 6916d059-c9a7-448d-9fad-0e3b0bb82a19

biemond, 2024-11-26:
Nice, looks ok

It seems for some you need a meter I guess.  So I guess we are done here. 
''''
totaldcpower 247 0 Total DC power
active_power_limit 80 -1 Nominal active power
temperature 298 -1 temperature
pvTodayEnergy 48 -1 daily energy yield
pvMonthEnergy 1226 -1 monthly energy yield
pvTotalEnergy 8443 -1 total energy yield
daily_export_from_pv 0 -1 daily export from pv
daily_direct_energy_consumption 0 -1 daily direct energy consumption
Client closed
''''

erikfriman, 2024-11-27:
Missing some flow cards though, specifically related to "The solar power changed" and "Total day yield changed"

biemond, 2024-11-27:
ok , will add

biemond, 2024-11-28:
now with flow triggers https://homey.app/a/solaredge.modbus/test/

erikfriman, 2024-11-29:
Thanks a bunch <3

biemond, 2024-11-29:
so let me check if I can add some flow /device actions

erikfriman, 2025-01-27:
Eveything is working like a charm now and this is more of a request and not an issue, would it be possible to get a "when" flow card for "battery power changed"? :)