biemond/solaredge.modbus #109: Solaredge firmware 4.21.xx and 4.22.xx breaks remote battery storage commands

thomaslehmann1234, 2024-11-24:
Hey folks,

I installed firmware 4.21.xx to my Solaredge SE10K-RWS and found out, that my HA remote storage commands stopped working after that upgrade, no effects from the remote battery commands. I enabled debug mode for the integration without any entries in the logs.

A test with firmware 4.22.xx resulted in the same issue. I requested a firmware rollback from Solaredge support to 4.20.xx and remote battery storage commands are back.

Solaredge startet to introduce new battery profiles in the latest releases so maybe modbus registers changed?

Any ideas?

biemond, 2024-11-25:
oh really, can you send me some diagnostic reports after these changes. 

plus their docs are public so their is some new version available. 
I know they support now profiles but that is mostly from the app. let me also search the internet for this

thomaslehmann1234, 2024-11-25:
Please find some debug output. One of my scripts is trying to execute remote storage commands. [debug.log](https://github.com/user-attachments/files/17907973/debug.log)
hth

biemond, 2024-11-26:
can you make some screenshots of all the inverter menu options.  maybe change by hand, capture the value of the registers

biemond, 2024-11-26:
only see failed to connect, saw on the docs they talk about lan only , not wifi and max 1 connection at a time. 
dont know if HA releases or reuse connection so you can do your write action. 
I dont see a modbus exception. 

in my app I do a poll every 30 sec , release the connection for other stuff like write, or keep on retrying.

thomaslehmann1234, 2024-11-26:
I am back on solaredge firmware 4.20.xx so i'm unable to screenshot. I am using lan, only one client and never had any connection issues. 

Anyone else upgraded to firmware 4.21.xx or higher?

biemond, 2024-11-26:
ok and now it works?

alecad, 2024-12-07:
I experienced the same issue after the update. However, the SolarEdge app now has a "Time of Use" feature ([more info](https://marketing.solaredge.com/tou-mode)) that allows me to manage my battery charging schedule according to my needs. As a result, I rely on the SolarEdge app for battery management and only use the Home Assistant integration for monitoring.

thomaslehmann1234, 2024-12-08:
Hi @alecad, could you please share your firmware version? I tried that "Time of Use" feature with firmware 4.21.xx and I was not successful with charing my battery from grid.

iNaiks, 2024-12-08:
> I am back on solaredge firmware 4.20.xx so i'm unable to screenshot. I am using lan, only one client and never had any connection issues. 
> 
> 
> 
> Anyone else upgraded to firmware 4.21.xx or higher? 

Hi, same issue with new firmware, how can I downgrade to 4.20?
I need to control how much is charging depending on the power of my house. 
Thanks

thomaslehmann1234, 2024-12-08:
Hi @iNaiks, Please open a case with solaredge support. They are able to rollback the firmware to 4.20.xx.

alecad, 2024-12-09:
> Hi @alecad, could you please share your firmware version? I tried that "Time of Use" feature with firmware 4.21.xx and I was not successful with charing my battery from grid.

Firmware version: 4.22.39. I also encountered issues setting up the “Time of Use” feature. After trying various approaches, I ended up performing a factory reset on the inverter and reconfiguring the parameters. I then had to contact technical support because the “import limit” option was no longer visible in SetApp. Unfortunately, I had to pretend to be an installer in order to speak with someone who fully understood the issue. They confirmed that there was a specific setting that isn’t configurable via SetApp, and they adjusted it on their end.
Now the “Time of Use” feature works properly. I was able to set an appropriate “import limit” to avoid any disconnections when charging the battery from the grid (or the EV), thus maximizing the draw without needing to modify my existing supply contract.

biemond, 2024-12-09:
do we think your reset was the solution or their fix on their end?

alecad, 2024-12-09:
> do we think your reset was the solution or their fix on their end?

The “Time of Use” feature didn’t work before the reset. After performing the factory reset, TOU started functioning, but I lost the “import limit” option within SetApp. SolarEdge support then had to intervene to restore that capability, as it’s not adjustable from the user side.
Also, something I’ve noticed lately is that TOU seems to operate on an hourly basis. In other words, if the battery is scheduled to charge from the grid, it will begin charging exactly on the hour (e.g., 5:00, 6:00) rather than starting at any time in between.

biemond, 2024-12-09:
this is all from the app or menu or not?  Do we already have some modbus docs about how to do TOU ( timing part) . I remember 2 years back you could do this on the site with an installer account and whole NL could see your TOU profile.

alecad, 2024-12-09:
> this is all from the app or menu or not? Do we already have some modbus docs about how to do TOU ( timing part) . I remember 2 years back you could do this on the site with an installer account and whole NL could see your TOU profile.

The Time of Use feature I’m referring to (available at https://marketing.solaredge.com/tou-mode) is entirely within the end-user’s monitoring app. It seems relatively new—at least it only appeared for me after the latest updates.

The installer-side setup you mentioned, where the installer creates a profile visible to everyone, is a different functionality and not what I’m talking about here.

According to their documentation, the end-user TOU functionality should charge the battery at specific times based on certain logic—such as average consumption and weather forecasts for the following day—rather than just following a simple, fixed schedule.

iNaiks, 2024-12-09:
In my case I'm using the "solaredge modbus for home assistant" and whit the update I can't control any limit on the battery.
I've tried some changes but I need to access with the wifi of the internet and access to local IP (172.16.0.1), but is very bad option to do it every time.
Now I'm waiting a response from solaredge to downgrade to earlier version (like @thomaslehmann1234 ) and return to use with home assistant.
If there is any new update that can work whit this firmware I would be grateful.
Thanks
Nico

biemond, 2024-12-09:
so I got version 4.21.23 , let me do some testing.

biemond, 2024-12-09:
this works for me  , what command did you guys try.
[image]
[image]

biemond, 2024-12-09:
but looks like I got 4kw import instead of the 2kw. did you also got something like that.?  let me try again.

I know it works from PV , I did this, whole my summer. 
[image]

biemond, 2024-12-09:
also discharge 500w works for me.  Maybe some HA issue?

thomaslehmann1234, 2024-12-09:
Thank you for checking @biemond. That is very interesting. Who is the maintainer of https://github.com/binsentsu/home-assistant-solaredge-modbus/? I raised the same issue in that project but never got an answer.

iNaiks, 2024-12-09:
I've tried some things and one worked 50%.
- Set "Energy Control" on setapp to "Time of Use + Remote" (before was only "remote" I think)
[image]

- Set the battery mode, Installator settings 
[image]

- Set "Storage control mode" to "Remote" on Home Assistant
And now the "Storage remote command mode" work for "Maximize self consumption" works, and the "number.solaredge_storage_remote_discharge_limit" works (in my case is entity with name "Bat. Discharge Limit".

But in "Charge from PV and AC" for charging the battery didn't work, the entity for limit charge "number.solaredge_storage_remote_charge_limit" with name "Bat. Charge Limit" always turns to 0. Same for "Maximize self consumption".
And I don't know for what is the entity "number.solaredge_storage_ac_charge_limit" because I change the value but nothing happens (now is set to 1500w but the battery didn't charge).

Maybe is needed to configure something on Time of Use?

[image]

iNaiks, 2024-12-09:
@thomaslehmann1234 is this working for you?

biemond, 2024-12-09:
4.21.23. https://www.solaredge.com/us/support/firmware-release-notes/residential-inverters-setapp

SolarEdge Home Battery 400V - AC charge - New definition | New definition for “Grid charge” 
feature:
1. “Disabled” - The battery will automatically charge from the grid only when the battery will be fully discharged (SOE<0%). 
2. “Maintenance” - The battery will charge from the grid only when the SOE<Backup Reserve, or when SOE<0%3. 
3. “Enabled” - No limitations for charging the battery from the grid. The battery will charge from the grid according to “Disabled”/“Maintenance” definitions, and defined TOU profiles.

```

biemond, 2024-12-09:
also had users where this was disabled somehow 

```
      if ( type== 'storageacchargepolicy') {
        // 0 – Disable
        // 1 – Always allowed – needed for AC coupling operation. Allows unlimited charging from the AC. When used with Maximize self-consumption, only excess power is used for charging (charging from the grid is not allowed).
        // 2 – Fixed Energy Limit – allows AC charging with a fixed yearly (Jan 1 to Dec 31) limit (needed for meeting ITC regulation in the US)
        // 3 – Percent of Production - allows AC charging with a % of system production year to date limit (needed for meeting ITC regulation in the US)
        const storageacchargepolicyRes = await client.writeSingleRegister(0xe005, Number(value));
        console.log('storageacchargepolicy', storageacchargepolicyRes)
      }
```

iNaiks, 2024-12-09:
Sorry, I don't know how to use from your repo. I only used the Home Assistant HACS integration from binsentsu. 
So I understand that this integration need to be updated for the new definition for 400v home battery. 

I will try to understand how your repo works and try to make a form from other repo. 

Very thanks for the info !!!

biemond, 2024-12-09:
yeah we use homey or better we use nodejs but I know on HA you can also do modbus read and write  , you can check register 0xe005 or 57349

or install nodejs , install the packages and run this js script on node. like this https://github.com/biemond/solaredge.modbus/blob/main/test.js

iNaiks, 2024-12-10:
I don't know what happen but this morning all is working fine 🤣, remote commands works for charging and now the "bat.charging limit" keeps the numer i set. I don't know

biemond, 2024-12-10:
yep solaredge night batch.  they do some stuff for batt during that time.  also had it on the 1st days. by default they disable it all, when used and it is ok they will enable it again.

iNaiks, 2024-12-10:
So they are changing settings to all people and are going to change it again?

iNaiks, 2024-12-12:
It seems that SolarEdge has dowgraded my inverter that night. 
With version 4.20.xx all works fine

biemond, 2024-12-12:
ack. but after reset and that night, it also worked for you on 4.21 or not?

iNaiks, 2024-12-13:
I think was the downgrade but I'm not 100% secure. 
The email of downgrade was the third day, but nothing has changed. 
The only change was the second day that all works good. 
Maybe SolarEdge has made the downgrade and reply to my case the next day.

peter-dolkens, 2024-12-15:
In the other thread I found this comment:

https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/259#issuecomment-2522778912

Stack trace looks like there's a signed/unsigned type issue with a previously unpopulated value, and the latest firmware is finally populating it.

My battery is getting installed on Wednesday, so watching closely.

Hope this helps

biemond, 2024-12-15:
can be or it is more a HA issue, curious which register it is then . at least. I don't have any issue with 4.21 and my 2 batteries.

```
    batt_registers: Object = {
        "c_manufacturer": [0xe100, 16, 'STRING', "Manufacturer"],
        "c_model": [0xe110, 16, 'STRING', "Model"],
        "c_version": [0xe120, 16, 'STRING', "Version"],
        "c_serialnumber": [0xe130, 16, 'STRING', "Serial"],
        "c_deviceaddress": [0xe140, 1, 'UINT16', "Modbus ID"],
        "c_sunspec_did": [0xe141, 1, 'UINT16', "SunSpec DID"],

        "rated_energy": [0xe142, 2, 'SEFLOAT', "Rated Energy"],
        "maximum_charge_continuous_power": [0xe144, 2, 'SEFLOAT', "Maximum Charge Continuous Power"],
        "maximum_discharge_continuous_power": [0xe146, 2, 'SEFLOAT', "Maximum Discharge Continuous Power"],
        "maximum_charge_peak_power": [0xe148, 2, 'SEFLOAT', "Maximum Charge Peak Power"],
        "maximum_discharge_peak_power": [0xe14a, 2, 'SEFLOAT', "Maximum Discharge Peak Power"],

        "average_temperature": [0xe16c, 2, 'SEFLOAT', "Average Temperature"],
        "maximum_temperature": [0xe16e, 2, 'SEFLOAT', "Maximum Temperature"],

        "instantaneous_voltage": [0xe170, 2, 'SEFLOAT', "Instantaneous Voltage"],
        "instantaneous_current": [0xe172, 2, 'SEFLOAT', "Instantaneous Current"],
        "instantaneous_power": [0xe174, 2, 'SEFLOAT', "Instantaneous Power"],

        "lifetime_export_energy_counter": [0xe176, 4, 'UINT64', "Total Exported Energy"],
        "lifetime_import_energy_counter": [0xe17A, 4, 'UINT64', "Total Imported Energy"],

        "maximum_energy": [0xe17e, 2, 'SEFLOAT', "Maximum Energy"],
        "available_energy": [0xe180, 2, 'SEFLOAT', "Available Energy"],

        "soh": [0xe182, 2, 'SEFLOAT', "State of Health [SOH)"],
        "soe": [0xe184, 2, 'SEFLOAT', "State of Energy [SOE)"],

        "status": [0xe186, 2, 'UINT32', "Status"],
        "status_internal": [0xe188, 2, 'UINT32', "Internal Status"],

        "event_log": [0xe18a, 2, 'UINT16', "Event Log"],
        "event_log_internal": [0xe192, 2, 'UINT16', "Internal Event Log"]
    }

```