biemond/solaredge.modbus #111: Sungrow Hybrid  Charge/Discharge

malanix, 2024-12-07:
The Sungrow integration works fine. Good work.

But when trying to use the Charge/Discharge card with command Charge, a parameter is supplied. I supposed that the card parameter would correspond to the 'Charging/Discharging Power' parameter in the Sungrow hybrid inverter. I was looking into the inverter parameters via the web interface. After issuing the Charge command with a parameter 5, and a refresh of the web parameters, the Charge was updated fine, but the 'Charging/Discharging Power' was set to 0.0; A Stop was issued. I was then setting 'Charging/Discharging Power' to a value via the web interface and issued the command again. As before the Charge went through, but the 'Charging/Discharging Power' was again set to 0.0.

I did this debugging after first finding that nothing happened. No current was drawn. And the inverter was doing what it was told to do charging with 0.0 kW. Setting the parameter 'Charging/Discharging Power' after that, made charging take place according to the value I had set.

Now I can have misunderstood something. Please enlight me.

Mats Nilsson, Täby, Sweden

biemond, 2024-12-07:
Hey,

I set these  but I have no sungrow.  maybe you can check with modbus app to see what works or not.
```

      if (type == 'export') {
        const exportLimitationRes = await client.writeSingleRegister(13086, command);
        console.log('exportLimitation', exportLimitationRes);

        const exportPowerRes = await client.writeSingleRegister(13073, value);
        console.log('exportPower', exportPowerRes);
      }

      if (type == 'charge') {
        const chargeCommandRes = await client.writeSingleRegister(13050, command);
        console.log('chargeCommand', chargeCommandRes);

        const chargePowerRes = await client.writeSingleRegister(13051, value);
        console.log('chargePower', chargePowerRes);
      }
```

malanix, 2024-12-08:
Testing according to your answer is beyond my capacity.

However looking into  Home Assistant implementation by Martin Kaiser for
 - name: Battery forced charge discharge power
# for my *SH10RT it is set in Watt*, not in % unit_of_measurement: W

I tried to set the parameter to 5000, and magic, 5000 W was drawn! So your
implementation works fine.

However hard to figure out that you should give the parameter in W not kW
as is documented looking into the Sungrow Hybrid Inverter.

So all is fine, thanks for the prompt answer.

Mats

biemond, 2024-12-08:
ok, can you check which values will work and which are not.   in your 1st comment you mentioned 0 , did you do 0 as power value?  or 5 for example

malanix, 2024-12-08:
Before i wanted 5 kW and set the value 5, believing the measurement was in
kW. This value was truncated to 0.00 kW as the values should be given in
Watts.

Now specifying 5000 gave 5.00 kW in the inverter.

biemond, 2024-12-08:
ok will update the param

malanix, 2024-12-08:
No I think not. There might be flows that stop working. Better if a comment
or description explains.

biemond, 2024-12-08:
yep that is what I did

biemond, 2024-12-09:
that is weird, last change was from early sept and only changed the timeouts https://github.com/biemond/solaredge.modbus/commit/af7e7cf9109c28fbf745d5aa1e9c3ff566318a89#diff-194bb2ef1812df1d87f5241d6e06371d1c6bb68379cf2ddc0612fed9a64dffec

 can you send me a diagnostic report.

biemond, 2024-12-09:
and updated a label https://github.com/biemond/solaredge.modbus/commit/9a90a84bb0a97b96753f75381f94207a854124e6#diff-818c1bc69b0017077fb123b3992933b52e3eddbb3c00b0c35683d6af8d9887b4

biemond, 2024-12-09:
I see 1 crash report about a wrong value ```Error: InvalidValue
    at new WriteSingleRegisterRequestBody (/app/node_modules/jsmodbus/dist/request/write-single-register.js:47:19)
    at ModbusTCPClient.writeSingleRegister (/app/node_modules/jsmodbus/dist/modbus-client.js:90:23)
    at Socket. (/app/drivers/sungrowhybirdwithbatt/device.js:130:53)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)```

johsands, 2024-12-09:
Created a diag report, below is the number for it
009fa3f2-ec28-4988-940c-4c750ea87fe5

biemond, 2024-12-09:
yep replied, invalid value I got back from your inverter

johsands, 2024-12-09:
Yeah, saw that my inverter didn’t change discharge value. Seems it also lost variables I had as input for discharge around the time the new version was applied. 
I then created a manual flow to test it manually instead of my fully automated flow.

johsands, 2024-12-09:
Hi again,
Can this be connected to the previous posts regarding using W or kW?
Because sending in 1000kW would be bad for the converter 😳

biemond, 2024-12-09:
it should be in W and this was only a label change

johsands, 2024-12-09:
Checked the changelog and basically it was some more describing text but then the changes on senson vs picker.

Not sure why this happens, I can try to remove and re-install again. best would be to be able to pick another version of the app but that haven't been fixed on homey yet :(

johsands, 2024-12-09:
Tried uninstall and re-install again. The issue seems to be with charge/discharge after the latest update.
When ever I try to manually switch Charge or discharge I get below error:
[image]

johsands, 2024-12-09:
I can however change it to self-consumption. So it seems the discharge/charge does not work as intended :(

biemond, 2024-12-10:
no change on my side and you can check the code.

johsands, 2024-12-11:
I have now the same issue as @malanix .
I called my solar company which went in as installer/administrator in my system and could see that the charge/discharge did not change and had also ended up looping. a thought was this is because switching from self-consumption to force charge/discharge had made that.
They set a fixed value of 1000W as charge/discharge and worked fine.
And now when I try to change this value nothing happens to it, it's always 1000W, regardless if I set discharge or charge. So the card with Carge/discharge is not setting values anymore on sungrow systems with battery.
I was able to set the export limit to a value that changed just to verify some items worked. But still, the charge/discharge is malfunctioning since monday somewher between 12-13.

biemond, 2024-12-11:
interesting make a support call on sungrow.  let me check the docs for this fixed value. there must be some register which is set

johsands, 2024-12-11:
There is a guy that makes the similar automation/configuration for HA, perhaps there are some clues in there?
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main

biemond, 2024-12-11:
```
      if (type == 'charge') {
        const chargeCommandRes = await client.writeSingleRegister(13050, command);
        console.log('chargeCommand', chargeCommandRes);

        const chargePowerRes = await client.writeSingleRegister(13051, value);
        console.log('chargePower', chargePowerRes);
      }

```

```
      - name: Battery forced charge discharge cmd raw
        unique_id: sg_battery_forced_charge_discharge_cmd_raw
        device_address: !secret sungrow_modbus_slave
        address: 13050 # reg 13051
        input_type: holding
        data_type: uint16
        precision: 0
        state_class: measurement
        scan_interval: 10

      - name: Battery forced charge discharge power
        unique_id: sg_battery_forced_charge_discharge_power
        device_address: !secret sungrow_modbus_slave
        address: 13051 # reg 13052
        input_type: holding
        data_type: uint16
        precision: 0
        # datasheet says:
        # 0 to 5000 W for SH*K-*
        # 0 to 100 % for SH*.0RT
        # for my SH10RT it is set in Watt, not in %
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scan_interval: 10

```

biemond, 2024-12-11:
# datasheet says:
        # 0 to 5000 W for SH*K-*
        # 0 to 100 % for SH*.0RT
        # for my SH10RT it is set in Watt, not in %

johsands, 2024-12-14:
Alright, done some testing now and cloned this repo and ran it as development.
Using latest version, the SET for charge/discharge does not set correctly, it basically take the old value and set's it again, resulting it that it never changes. My initial issue was that I successfully set it to zero and then it stopped working with new version.

When I revert the repo to previous release(git reset --hard c6d84d3) it works perfectly and set the charge/discharge as intended.

If you cannot find the issue in the code can you please revert to c6d84d3 since it's broke sungrow?
saw that there is an additional issue logged on this exact same item.

biemond, 2024-12-14:
thanks stupid me , want to change the label and not the parameter name https://github.com/biemond/solaredge.modbus/blob/main/drivers/sungrowhybirdwithbatt/driver.flow.compose.json#L172

will revert it

johsands, 2024-12-14:
Awesome, thank you!

biemond, 2024-12-14:
here we go https://homey.app/a/solaredge.modbus/test/

johsands, 2024-12-14:
Great, Installed it through the homey app webpage since it was a test.
Tested out setting charge/discharge and it worked again!