biemond/solaredge.modbus #112: Sungrow and Homey Integration doesn't work

Swcu38, 2024-12-14:
Hello, i installed Homey Pro in order to use Tibber and my Solarplant in one application. My Sungrow Sh10Rt is not sending any kind of information to the homey pro. Can anyone help me? Thx

biemond, 2024-12-14:
hey , 

check if the port is alive and ip is ok , also this https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/doc/help.md

Swcu38, 2024-12-15:
Hey Edwin,Thanky you very much. Now i can See all Informations in the homey pro After Connecticut the inverter by cable. But there is still a Issue and i am not Sure if do somthing wrong. Is there a manual for creating flows? I create a flow with tibber. When current Price is under XX€ than charge Battery... but the inverter doesn't Start charging.Mit freundlichen GrüßenSascha Weidmann