biemond/solaredge.modbus #113: Huawei EMMA not working with homey pro

wachaudhary, 2024-12-15:
Hi, I am trying to connect Homey pro with my Huawei inverter and battery using Huawei EMMA. After giving the right IP address, data and information is not being pulled from EMMA and unable to see any data in the app

biemond, 2024-12-15:
hey can you send me a diagnostic report.

wachaudhary, 2024-12-15:
> hey can you send me a diagnostic report.

How do I extract the Diagnostic report ? :)

biemond, 2024-12-15:
go to the solaredge app , configure , send diagnostic report.

biemond, 2024-12-15:
also did you follow the forums , I think you need to disable tls certificate

biemond, 2024-12-15:
and enable modbus off course.

biemond, 2024-12-15:
at least  Alex Schatalow  and Fabian_E got it working. also you can check the HomeAssistance forums

wachaudhary, 2024-12-15:
> also did you follow the forums , I think you need to disable tls certificate

No I have not seen that forum. Please provide link on that forum :-)

wachaudhary, 2024-12-15:
[image]

I just verified ModBus-TCP is activated already

biemond, 2024-12-15:
https://github.com/wlcrs/huawei_solar/issues/498#issuecomment-2266771661
disable tls for modbus.

biemond, 2024-12-15:
I see some data in your report?  but you are not using the emma device 

```
[Driver:huaweiwithbatt] [Device:3cc3197a-a99b-4b1d-b1f2-38e1d051e1bc] 192.168.0.159
```

wachaudhary, 2024-12-15:
> I see some data in your report? but you are not using the emma device
> 
> ```
> [Driver:huaweiwithbatt] [Device:3cc3197a-a99b-4b1d-b1f2-38e1d051e1bc] 192.168.0.159
> ```

I have both devices (S-Dongle and EMMA). But wanted to verify if it is only EMMA which is not working or both are not working. I will send EMMA diagnostic soon.

wachaudhary, 2024-12-15:
EMMA diagnostic sent.  After disabling the TLS, still not data on Homey pro

[image]

wachaudhary, 2024-12-30:
After acting on all of your advices, I am still unable to connect EMMA to Homey pro.  TCP modbus is enabled and TLS is disabled as requested. Still no data information is updated in homey pro. :( 
P.S I connected dongle to inverter and it pulls the data correctly.

biemond, 2024-12-30:
did you do your networking correctly , 1 person had that , you have to do it a certain way.
else reach out to  Fabian_E on homey forum.

konradhuebner, 2025-01-18:
@wachaudhary  I just set that up, and it is working. You must enable Modbus and disable TLS by directly connecting to the EMMA system via the installer WiFi.  In the Homey app, set the inverter ID to 0 in the settings. After that, all worked for me immediately