biemond/solaredge.modbus #115: Sygenergy

goldstrike55, 2024-12-17:
How to connect Sigenergy battery?
I can't get data

goldstrike55, 2024-12-17:
[image]

I can't enable because it is connected to frank energy

biemond, 2024-12-17:
ah ok. mostly it should not block the EMS unless Frank made custom firmware.

biemond, 2024-12-17:
your installer can do that only https://community.home-assistant.io/t/sig-energy-system-integration/760448/14

goldstrike55, 2024-12-17:
Okay I will contact frank fore this thenxs

biemond, 2024-12-17:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/wiki/1.-Prerequisites

goldstrike55, 2024-12-17:
Okay I will ask my installateur thenxs

goldstrike55, 2024-12-18:
it works nou thenxs fore the support

goldstrike55, 2024-12-18:
[image]

In Homey on the energy tap it doesn't show the battery only the solar panels. How can I fix this?

goldstrike55, 2024-12-18:
Okay then I will wait some time and hope It will be fixed thenxs

biemond, 2024-12-18:
yes

goldstrike55, 2025-03-13:
Do you know when there is a fix to show the battery ?

biemond, 2025-03-13:
still beta , I think they are working on it.

goldstrike55, 2025-05-27:
Is there already a fix so I can see the battery status in anergy tap?