biemond/solaredge.modbus #116: Connection to sungrow inverter stops

frean087, 2025-01-08:
Hi,
The integration to sungrow inverter SH15T stops working after around two days. Restarting the app solves the issue.

Please find dump here:

9b3b4dd2-4a0e-4a4b-b09e-9b5d973dcd6a

Thanks for making this possible, hope there is an easy fix.

frean087, 2025-01-10:
Yes, homey is the only device that pulls data.
We setup an advance flow in homey that decide if the inverter should charge the battery connected or not. But the flow only checks on an hourly basis if the new hour is the cheapest one and then decides if charging is needed or not.
Additional info: Yeasterday I got an new update for the inverter since then it's been stable, needs to wait some more day in order to see if that solved it.
Thanks for looking into this.

biemond, 2025-01-10:
cool,  did not hear this from others , else when it is stuck again, send me a diagnostic report. 

maybe it is a homey  or app issue or some network.  can you check the mem and cpu usage of the app on insight.