biemond/solaredge.modbus #119: Solar Power not shown in graph

Haandrwalt, 2025-01-23:
After migrating to the new Homey Pro 2023, the graphs for “Solar Power” (SolarEdge Inverter, connected via Modbus) do not work anymore (“No data”). Only for this specific attribute there is no data for the graph since the migration 2 days ago. When extending the range from a day to a week, I do see data / a graph for the days before migrating to the Pro ‘23. I also see recent data for other attributes, like “Power Used” and “Total Yield”. Only for the “Solar Power” attribute, data cannot be shown in a graph anymore. 
The device tile shows the Solar Power as status indicator. That value constantly moves and confirms data is available, but for the related graph, there is “no data” available. 

Very strange and I cannot figure out what is going on. I have Modbus connection that works fine for years. Now, related to migration to new Homey Pro 2023, everything works, EXCEPT for the Solar Power graphs of my SolarEdge inverter. Graphs for all other attributes are not impacted, only the “Solar Power” data. 

[image]

[image]

[image]

biemond, 2025-01-23:
you mean on insights?  new homey 2023 does not honor my naming attribute , I think it is now called power usage

[image]

Haandrwalt, 2025-01-23:
Yes on Insights, Solar Power is shown as available attribute, but no data from Pro 2023 is available.

[image]

biemond, 2025-01-23:
it is power usage, the new name on pro 2023 ( or original name of measure_power) . it is a homey bug. I reported it to homey but they did not fix it.

Haandrwalt, 2025-01-23:
So you mean it is just a naming thing, where Old Homey used “Solar Power”, that is now called “Power usage” and the old attribute data is now available under “power usage” as well?

closed 2025-01-23 by Haandrwalt