biemond/solaredge.modbus #120: Solis inverter + battery

oelen000, 2025-01-27:
Hi, is it possible to controll a Solis hybrid inverter with battery in this app?

biemond, 2025-01-27:
yeah that is possible, 

does it allow modbus tcp, some doc with defintions and do you have a vpn so I can test it against.

oelen000, 2025-01-30:
Hi Edwin,

Volgens mij kan ik verder in het Nederlands. Sorry voor mijn late reactie.
De datalogger zou modus over IP moeten hebben volgens de documentatie.

https://solis-service.solisinverters.com/nl/support/solutions/articles/44002530087-solis-s2-wl-st-modbus-tcp-communicatie
https://usservice.solisinverters.com/support/solutions/articles/73000560403-solis-modbus-protocols-maps-

Ik heb een VPN aan laten maken.

[IMG_2139.PNG]

ipadres van de logger is 192.168.10.61
Admin
Solute0922!

Je hulp wordt erg op prijs gesteld

Met vriendelijke groet,

Mike Oelen

A  Weidebloemenweg 45, 9302 AC Roden

T 050-7119319  <tel:050-7119319>

M 06-42908015  <tel:06-42908015>

E ***@***.***  ***@***.***>

W www.tstroom.nl  <http://www.tstroom.nl/>

IMPORTANT: The contents of this email and any attachments are confidential. They are intended for the named recipient(s) only. If you have received this email by mistake, please notify the sender immediately and do not disclose the contents to anyone or make copies thereof.

biemond, 2025-01-30:
will check it in the weekend, also removed your details. this is on github.

biemond, 2025-02-01:
mail even naar biemond at gmail dot com

ik mis je vpn gegevens