biemond/solaredge.modbus #122: Solaredge total day yield issue

seb1300, 2025-02-01:
Hello,
I'm having an issue with the total day yield value. E.g. i Can read 251kwh instead of  11,2kwh, the Real value from Solar edge app.
This value doesn't correspond to the total yield neither.
I've noticed that issue with the New energy tab on homey.
Thanks you in advance for the job your are doing.

biemond, 2025-02-01:
hey 

do you also have a battery?  daily on solaredge is calculated it does not exists as register . I use meter_power for this. (  "energy_total": [0x9c9d, 2, 'ACC32', "Total Energy"],)

you can also try this if that helps https://homey.app/en-nl/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/test/

seb1300, 2025-02-01:
I tried the New app you mentionned but no improvement.

I don't have any battery