biemond/solaredge.modbus #123: Huawei inverter

valesneriq, 2025-02-02:
My flows has stopped working. Has the when card"power changed" dissapered?
Tried reinstall the app and tried adding it again but cant se this card.

valesneriq, 2025-02-02:
I have auto update on running ver 12.3.0 on my homey

did you do a homey upgrade?

biemond, 2025-02-02:
from  12.0 they removed some default triggers,  I think there should be a new 1 for measure_power, like solar changed or so

valesneriq, 2025-02-02:
No rhere is only more or less cards 😭

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

biemond, 2025-02-02:
will add it

valesneriq, 2025-02-02:
Good. Thank you!👍👍

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

szferi91, 2025-02-09:
Hello, any plan to add values from the power meter? something similar to what is in the HA (https://github.com/wlcrs/huawei_solar). the active power would be useful as if the value is positive electricity is fed to the grid, otherwise is pulled from the grid.

[image]

biemond, 2025-02-09:
I added inverter and meter active power ,is this not ok for you.

        "ACTIVE_POWER": [32080, 2, 'INT32', "ACTIVE_POWER", 0],    -> measure_power.active_power
        "POWER_METER_ACTIVE_POWER": [37113, 2, 'INT32', "POWER METER_ACTIVE_POWER", 0],   ->

measure_power.grid_active_power

szferi91, 2025-02-09:
hello, is that added also to the 2 inverter version? I'm using that and I don't have this information there. I'm using v 3.0.173

szferi91, 2025-02-09:
for instance, I have these values for the power meter

[image]

for the inverter I have 

[image]

biemond, 2025-02-09:
let me check , mostly do meters with battery , even when you don't have a battery.

biemond, 2025-02-09:
indeed on double it is not , let me add it.

szferi91, 2025-02-09:
I think those values are taken from the inverters, if we add them up we get to the total production value.

[image]

the value of the power meter shows the difference between the consumption and the total production. that one is not available currently in the app, only the active power from the inverters

biemond, 2025-02-09:
it is this        

 "ACTIVE_POWER": [

[image]

, 2, 'INT32', "ACTIVE_POWER", 0],   

       "POWER_METER_ACTIVE_POWER": [37113, 2, 'INT32', "POWER METER_ACTIVE_POWER", 0],  

https://github.com/wlcrs/huawei_solar/discussions/826

szferi91, 2025-02-09:
[image]

[image]

biemond, 2025-02-09:
Can you send me diagnostic report with Huawei double

szferi91, 2025-02-09:
sensor.power_meter_active_power -> This is the entity ID in HA I'm referring to

szferi91, 2025-02-09:
the requested diagnostic report: 75d9b51e-84cf-4c7a-be0e-d4f843dd7075

[image]

active power is negative 

[image]

only positive values in homey:

[image]

biemond, 2025-02-09:
1st with s-dongle don't do HA and homey together , remove your devices and only keep 1 and disable HA, s-dongle is cheap hardware which cannot handle 2 or more connections.  Go for emma but it will be 300 euro instead.

plus use only huawei with batt or double else you won't get meter values.

szferi91, 2025-02-09:
Yes, I have disabled HA, when it was used in homey. Currently, I have disabled the app in Homey, so I can read the correct data for both inverters and the Power meter, from the HA. I will test it again when I will have also a battery added to my system later this year.
Thank you for your time and effort

biemond, 2025-02-09:
cool but should also work without a battery,  let me know.  the app should be handle to handle it.  
mostly meters you have only for battery or export restrictions.  that is why I added it under battery device.

szferi91, 2025-02-09:
It doesn't work unfortunately without a better, that's why I asked you today. I installed HA only today to check if it works there or not. I have checked it with simple inverter inverter with battery and also the 2 inverter version. I have tried then in HA and it returned the values. Will get back to you when I will also have battery in my system. Thanks again for your feedback, and enjoy the rest of your weekend

biemond, 2025-02-09:
ok , disable HA , remove all homey Huawei devices , wait 15 min and add with battery until you get some values. 
then you can send me a report, like to fix it .

but normal device won't show it indeed.

szferi91, 2025-02-09:
should I tick the checkbox for "has battery" to have the correct values?

biemond, 2025-02-09:
```

  let finalRes = { ...checkRegisterRes }
      if (this.getSetting('has_battery') == true){
        finalRes = { ...checkRegisterRes, ...checkBatteryRes, ...checkMetersRes }
      }  else {
        this.removeCapability('battery');
        this.removeCapability('measure_battery');     
        this.removeCapability('measure_power.grid_active_power'); 
        this.removeCapability('measure_power.batt_charge');
        this.removeCapability('measure_power.batt_discharge');   
        this.removeCapability('measure_power.grid_phase1');
        this.removeCapability('measure_power.grid_phase2');   
        this.removeCapability('measure_power.grid_phase3');
```

szferi91, 2025-02-09:
this was the trick, so if none of them had batter the grid_active_power, was removed. it makes sense now. Thank you for your help again, now I have the information in the homey app also, I just have to sum them and I will have the real value

biemond, 2025-02-09:
awesome , enjoy

also need to change this part because not everyone has 3 phase grid connection and you will get some weird big number for the other phases

szferi91, 2025-02-09:
let me know if I can help with anything

biemond, 2025-11-03:
yeah it is meters with battery.  and in your case no batt info then.