biemond/solaredge.modbus #125: Request: Add monthly and yearly yield SolarEdge Inverter

SkorpionNL, 2025-02-17:
Hi Edwin,

I would like to request an addition of the monthly and yearly yield for the SolarEdge inverter.

Thanks!

biemond, 2025-02-18:
hey,

that is hard because I need to calculate it and reset it on the end of the month or year.  it can only go wrong. 
it is probably it is easier to do this with a flow and store a value at the beginning of the month.

SkorpionNL, 2025-02-18:
Thanks for your quick reply!

That’s unfortunate. I was hoping that SolarEdge could deliver those numbers via their API and that it would have been an easy pass through.

In this case this indeed isn’t worth the effort.

I will solve it via flows like you mentioned.

Thanks anyway and keep up the good work!

biemond, 2025-02-18:
yeah can check their website api, they also calculate this.   but on the inverter side I have only total.