biemond/solaredge.modbus #126: Growatt SPH issues

voroniys, 2025-02-24:
I've just started to use this app, but have found several issues which makes it less usable.
First - one that should be very easy to fix: when I look into the tags list in the Logic app there are two tags called 'Battery' 
Please rename the tags, so they are clearly indicates what it is, like 'Battery SOC'

Now more complex stuff :)
I'm using Growatt SPH5000 TL3 BH (not UP) and at least with regards to this invertor I've found the following issues:

1. Register 1044 is read only, so it can be used for writing to enforce a Priority. In SPH there are 3 timeslots for 'Grid first' and 3 time slots for 'Batt First'. If they are all disabled or current time is not matching any of the slots - Priority is 'Load First' If there is a matching slot - priority set to either 'Grid First' or 'Battery First'. I have no idea what would happen if 2 slots of different priorities will match the same time. 

So to enforce 'Load First' value zero has to be written into registers 1082, 1085, 1088, 1102, 1105, 1108 - i.e. disable all 6 timeslots for other priorities.
To enforce 'Battery First' or 'Grid First' it is possible to set not exposed in the app slot, for instance slot#3 to 00:00-23:59 and Enable (registers 1086-1088 for Grid First and 1106-1108 for Batt First) . To avoid any issues all other slots set to disable. 

2. The SPH invertor has 2 different Minimum SOC settings - one for Grid First priority - register 1071 which is used in the app currently, but there is also different Minimum SOC for the priority Load First. This minimum SOC is in the register 608 and not used in the app at all. 

IMHO the best would be to create 2 minimum SOCs properties in the app - one called 'Grid First minimum SOC', another 'Load First minimum SOC' and create Flow action cards to set each of them.

If you need any additional information I would be happy to provide it as well as to help with development/testing.

biemond, 2025-02-24:
hey,  I don't have a growatt , it was a favor for someone to do the same as solaredge and try to do the minimal of the minimal.

there are 2 battery caps because they are shown on different spots, like sensor and battery tab so that is why. more for layout , should have the same value.

1, is not my problem, I expect user knows what to do, this app is just 1 on 1 to modbus , no app logic or protection . you need to find out what works or not. Also I expect you know what you do and if you set some things in a different app you need to go the defaults.

2 did not have this request before  but should be easy to implement. 

Welcome to send a PR , it is all open source else I can add 2 . what is the device do you use  "Growatt + Battery" or "Growatt TL-X + Battery"?

thanks Edwin

biemond, 2025-02-24:
there is also now a modbus app where you can set other things I currently does not support.

biemond, 2025-02-24:
cool,  for 1 we have to disable 1st all the profiles and then enable only 1.
but if you can do this and test it all , it will be the best because you own 1 and use it on daily basis.  I have no clue how it works.

I will add 608 and make a flow action for it if it works like 1071

voroniys, 2025-02-24:
I've createated a PR for 1. Please let me know what do you think.

voroniys, 2025-02-24:
> 
> I will add 608 and make a flow action for it if it works like 1071

Yes, 608 works exactly the same like 1071 - the value in the register is the stop discharging SOC for corresponded priority and it is writable.

biemond, 2025-03-01:
so 608 is this  , not like 1071. is this what you want?

[image]

biemond, 2025-03-01:
also see this 

```
 - platform: modbus_controller
    modbus_controller_id: modbus_growatt
    name: "SPH Min SOC"
    id: sphminsoc
    register_type: holding
    address: 608
    unit_of_measurement: "%"
    value_type: U_WORD
    accuracy_decimals: 0
```

biemond, 2025-03-01:
also these are valid sph registers, do you a better doc for me 

[image]

voroniys, 2025-03-02:
> also see this
> 
> ```
>  - platform: modbus_controller
>     modbus_controller_id: modbus_growatt
>     name: "SPH Min SOC"
>     id: sphminsoc
>     register_type: holding
>     address: 608
>     unit_of_measurement: "%"
>     value_type: U_WORD
>     accuracy_decimals: 0
> ```

I was looking for this setting of SPH, but failed to find it in any kind of documentation. But during my search I have seen somewhere something like this, so I've tested it by setting the different SOC percentages using Growatt's website and have seen it is indeed changing the value of register 608. Also tested it with ModBus tester by setting a value in this register and indeed in Load First mode battery discharge stops at the value from this register.

So, I don't have any documentation regarding this register it is a kind of reverse engineering effort. But it is really working, tested many times.

So, to summarize:
1071 - minimum SOC at Grid First priority
608 - minimum SOC at Load First priority

biemond, 2025-03-02:
nice, let me add it

biemond, 2025-03-02:
can you try it out https://homey.app/a/solaredge.modbus/test/

voroniys, 2025-03-03:
Thanks a lot! I've created several workflows to test most of the features we've added. Let me day or two for testing, I'll let you know the results

biemond, 2025-03-03:
cool so the flow actions are working at least?  heard from another sph user it works for him which reported same issue at the same time as you

biemond, 2025-03-03:
will do a rename of the label on the other soc

voroniys, 2025-03-04:
My testing is not that good, as I hoped. 

First the good thing - register 608 stuff working perfectly, no issues there.

But two another changes - time sync and priority change are not working as expected. 

Time sync is always successful update seconds, but the rest may vary, year is almost never updated properly. I suspect writing multiple registers very quick causing some issues. I don't know what to blame - Growatt or Waveshare modbus gateway. Will do some more testing. For time it should be possible to use write multiple registers, but this causes errors and does not work with modbus tester I'm using, so I can't use it.

And when I update single register with the tester it takes several seconds for me to update next register and everything works as expected. But Homey App doing it all in miliseconds and this probably the reason.

Enforcing priority also worked not as expected - I've enabled several periods for both BF & GF, but when I executed Flow with enforcing LF it dropped somehow to 0 not only registers that enable/disable the periods, but also the start/stop times. I still blame to fast write actions.

So I need to do more testing. I'm going to create my own very simple app to learn stuff and to test different ways of writing registers. I'll let you know when I get more information. 

The issues could be caused also by modbus gateway I'm using. So more testing is needed

biemond, 2025-03-04:
can you send me a diagnostic report so I can see it ,  or install the app locally  with homey app run --remote you will see the logs then

biemond, 2025-03-04:
we used await on it or not?

biemond, 2025-03-04:
``` delay(ms: any) {
    return new Promise(resolve => setTimeout(resolve, ms));
  }

      await this.delay(500);```

voroniys, 2025-03-23:
Hi Edwin, 
I've played a bit with eslint for my own educational purposes, here is the result:
https://github.com/biemond/solaredge.modbus/compare/main...voroniys:solaredge.modbus:lint?expand=1
Number of issues is reduced from 16k+ to 4k+ (eslint --fix), All issues in growattwithbatt solved.
Do you think it is worth of creating a PR ?

biemond, 2025-03-23:
yes , awesome . very welcome.

closed 2025-03-27 by voroniys

biemond, 2025-05-25:
hey 

got a question for you for normal growatt with no meters installed.   Does your export limit / enabled works without meters?
and can you test registers  00 (on/off)  and 03 (active power)  , it seems people like to reduce output with negative prices

[image]

thanks

voroniys, 2025-06-17:
Hi, sorry haven't see your message. I've tested Reg 0 - it works. It is possible to switch inverter off, and then back on. I you you wish I can create a PR with this action.
I can't test how export limit is working without the meter, but IMHO it is essential. What I see on my inverter it constantly gets measurements from the meter and adjust inverter to be near zero, although at least mine always stays a bit negative - approx 20-80 W export with the limit set to 0%. Same is in Load First mode - it keeps export to the grid around 20-80 W. I have not found any methods to influence this behaviour.