biemond/solaredge.modbus #129: Sungrow Mode ”Self-consumption mode”

MartinKarlg, 2025-02-28:
From later this day the card ”self consumption mode” charges the battery. At the moment i cant selfuse the battery. Only charge/discharge it.

biemond, 2025-03-01:
no clue , nothing has changed for months, what does your inverter menu say? did you got some update

MartinKarlg, 2025-03-01:
Meny says self-consumption and auto-update. Im not admin over my own plant. Installer company is. I cant find or see if i got an update yesterday…. Everything worked fine before.

biemond, 2025-03-01:
yeah become admin,  there are more issues with sungrow after some firmware upgrade https://github.com/biemond/solaredge.modbus/issues/124

MartinKarlg, 2025-03-08:
another question. why isnt self-consumption visible in this view? :) Is it possible to fix?

[image]

biemond, 2025-03-08:
probably the modbus spec does not allow this.  where did you see this in the docs?

MartinKarlg, 2025-07-26:
> Self consumption is in "EMS Mode", not "charge/discharge" command.
> 
> I'm not certain but quite often Sungrow will not switch between charge/discharge unless you stop before the change. Need to test it a bit more to make sure.

Now my system works fine. I found a solution to only use charge/discharge. Instead of self-consumption i use discharge. It works with information from my Pulse from Tibber.