biemond/solaredge.modbus #134: Huawei Battery in Energy Tab

tivb, 2025-03-19:
Hello!

Many thanks for creating this integration over modbus. Love the details that are visible for my Huawei Inverter and Battery!

My Home Battery is not showing up in the Homey Energy tab, which causes weird numbers in solar production. When my battery is discharging at night, this is added in the solar production totals in the energy tab, because homey is not really aware this is a home battery and it appears to think it is comming from the solar panels.

Thanks! 

Kr,
Tim

tivb, 2025-03-22:
Looking at the Homey Developper pages, it does indeed look that a seperate device within Homey is needed with the device class of battery and a driver with the homebattery property. 

https://apps.developer.homey.app/the-basics/devices/energy#home-batteries 

Not sure if this would be a lot of work as I'm not really a developper.

Olive974, 2025-04-09:
Bonjour d’abords merci pour l’appli.
est ce que la solution ne résiderai pas dans la creation par l’appli d’un virtual device type batterie par lappli en plus, qui serait alimenté en données par le flux mod us ?