biemond/solaredge.modbus #135: Solax Ultra - missing/incorrect parameters

JorgenSeemann, 2025-03-24:
Hello,

Would it be possible to add support for the Solax Ultra (30) model?
These has 3 PV (versus normal 2), and some of the parameters looks to be wrong / used differently.
Solar + batt power: Constantly 0, regardless of PV & house load
House load and grid in/output are looking suspiciously.

[image]

Cheers
Jørgen

JorgenSeemann, 2025-03-25:
> hey, do you have official modbus solax register doc?

Hello Edwin.

I got this from the support 😀

Best regards
Jørgen

biemond, 2025-03-25:
can you share it, I don't see it

JorgenSeemann, 2025-03-25:
Here we go :)

[(Solax)Hybrid X3-ULTRA Modbus RTU V1.00-English.pdf](https://github.com/user-attachments/files/19446525/Solax.Hybrid.X3-ULTRA.Modbus.RTU.V1.00-English.pdf)

biemond, 2025-03-25:
[image]

I don't see  total pv power without batt charge.  weird.

JorgenSeemann, 2025-03-25:
> <img alt="Image" width="1028" src="https://private-user-images.githubusercontent.com/1085004/426617943-b4a1e13d-dd2e-42d6-820b-176df6c77a2a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI5MjAzMjksIm5iZiI6MTc0MjkyMDAyOSwicGF0aCI6Ii8xMDg1MDA0LzQyNjYxNzk0My1iNGExZTEzZC1kZDJlLTQyZDYtODIwYi0xNzZkZjZjNzdhMmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMjVUMTYyNzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDM4OTE5MDYxNmYzOTc1ZTFlMmNmNWQxNmI1NTE0Yzk0YzZjZmY0ZjA4Y2U2NmRhOTU0N2JjMWRhMmQ2NjQ0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Pk60fWxz0xVcQFrfXMOq0Qdvv1Vlghr8YajUOIqbdWw">
> I don't see total pv power without batt charge. weird.

Yes that is wierd :/

JorgenSeemann, 2025-03-25:
> so do you have another solax inverter [#70 (comment)](https://github.com/biemond/solaredge.modbus/issues/70#issuecomment-2041542391) where it works?

Yes it works for my Solax X3-15, which only have 2*PV and 1*Battery

biemond, 2025-03-25:
so is this inverter a new generation? and if we call it solax ultra is that ok and everyone knows it then?

JorgenSeemann, 2025-03-25:
> so is this inverter a new generation? and if we call it solax ultra is that ok and everyone knows it then?

I think calling it Solax Ultra should be fine. It's a complete new model/type :)

JorgenSeemann, 2025-04-01:
> <img alt="Image" width="1028" [...]

I think they must be calculating the PV, Grid, Battery and house load.

PV generation is sum:  
0x000A Powerdc1 PV1 Power 1W uint16 1
0x000B Powerdc2 PV2 Power 1W uint16 1
0x0124 Powerdc3 PV3 Power 1W uint16 1

Grid Power is sum:
0x006C GridPower(R) On-Grid Power 1W int16 1
0x0070 GridPower(S) On-Grid Power 1W int16 1
0x0074 GridPower(T) On-Grid Power 1W int16 1

Battery Power is sum
0x0016 Batpower_Charge1 Batpower_Charge1 1W int16 1
0x0129 Batpower_Charge2 Batpower_Charge2 1W int16 1

Then house load would be sum(PV generation, Grid power, Battery power) ?

Make sense ?

biemond, 2025-04-01:
yes and solar/batt would be  pv1,2,3 - batt charge 1,2 + discharge1,2

JorgenSeemann, 2025-04-01:
> yes and solar/batt would be pv1,2,3 - batt charge 1,2 + discharge1,2

Hum... Don't seems to be able to locate discharge1,2 ?
There is 0x0019 System_BDCStatus (0: discharge, 1: charge, 2: stop) - uint16 1 - might be used if the Batpower_Charge1,2 cannot be negative (eq. Batpower_Charge1,2 is positive if System_BDCStatus is 0, and negative if System_BDCStatus is 1)

biemond, 2025-04-01:
yeah it is pos and negative indeed

JorgenSeemann, 2025-04-09:
> yeah it is pos and negative indeed

Do you need anything from me to be able to compile a version with support for Solax Ultra ?

biemond, 2025-04-09:
no, only need some time unless you can make this  device like the other solax.

JorgenSeemann, 2025-04-10:
> no, only need some time unless you can make this device like the other solax.

Great :)
Unfortunately I cannot alter the fw of the Solax, and I have no clue as to why they have implemented these changes. Beats me TBH :/

biemond, 2025-05-05:
let me  add a new device so you can test it.

JorgenSeemann, 2025-05-05:
Yes, please 😀

Med venlig hilsen
Jørgen

biemond, 2025-05-05:
can you try this https://homey.app/a/solaredge.modbus/test/

JorgenSeemann, 2025-05-05:
Hi Edwin

I get PV1-3 now (no sun so all 0), but compared to my other Solax I seem to
be missing: house load, grid in/out, temperature, meter
current/voltage/power (pr phase) ?

Cheers
Jørgen

Den man. 5. maj 2025 kl. 19.31 skrev Edwin Biemond ***@***.***

>:

>
> can you try this https://homey.app/a/solaredge.modbus/test/

biemond, 2025-05-05:
yep , most are gone with ultra ,  only load I can add. 

maybe you can check the doc.

biemond, 2025-05-05:
ok found temperature,  looks like the doc I got from you is not complete.

biemond, 2025-05-06:
how is pvpower going? 

did you find these registers? like   house load, grid in/out, meter
current/voltage/power (pr phase)

JorgenSeemann, 2025-05-06:
Well, grid seems to be on page 18-19: (0x04)
[image: image.png]
[image: image.png]
I guess they must now be calculating house load - cannot find it anywhere.
Like house load = PV1-3 + Battery1-2 - Grid R-S-T ?
(I have a vague memory of us doing similar on the old Solax?)
Grid current/voltage/power would be great. The Meter1+2 (page 22) seems to
be only giving today/total import/export. This seems to be present already
:)

PV1 works fine. I'll try PV2+3 over the weekend, as I'm currently having
only 1 string attached :)

Cheers
Jørgen

biemond, 2025-05-06:
can you check this https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solax.py is yours gen5?

JorgenSeemann, 2025-05-07:
yes my sn falls into this : ("H3BC", 60), # Gen5 X3 Ultra C :)

biemond, 2025-05-07:
can you check the above gen5 stuff which is interesting for you.

JorgenSeemann, 2025-05-07:
you mean like this ?

    SolaxModbusSelectEntityDescription(
        name="Charger Use Mode",
        key="charger_use_mode",
        register=0x1F,
        option_dict={
            0: "Self Use Mode",
            1: "Feedin Priority",
            2: "Back Up Mode",
            3: "Manual Mode",
            4: "PeakShaving",
            5: "Smart Schedule",
        },
        allowedtypes=AC | HYBRID | GEN4 | GEN5,
        icon="mdi:dip-switch",

Den ons. 7. maj 2025 kl. 19.06 skrev Edwin Biemond ***@***.***

>:

>
> can you check the above gen5 stuff which is interesting for you.

biemond, 2025-05-07:
yep but this entry is about a charger

biemond, 2025-05-07:
more like your missing load or meter params which I cannot find in your doc. 

I found temp which was also not on your pdf

JorgenSeemann, 2025-05-07:
Right. I located the below interesting GEN5 (personally I don't need all,
but would be nice for general support to include them).
I can see that they indeed calculate house load (right at the end of the
list here :)

    SolaxModbusSelectEntityDescription(
        name="Charger Use Mode",
        key="charger_use_mode",
        register=0x1F,
        option_dict={
            0: "Self Use Mode",
            1: "Feedin Priority",
            2: "Back Up Mode",
            3: "Manual Mode",
            4: "PeakShaving",
            5: "Smart Schedule",
        },
        allowedtypes=AC | HYBRID | GEN4 | GEN5,
        icon="mdi:dip-switch",
    ),

    SolaxModbusSelectEntityDescription(
        name="Device Lock",
        key="device_lock",
        register=0xB5,
        option_dict={
            0: "Unlock",
            1: "Lock",
        },
        allowedtypes=AC | HYBRID | GEN4 | GEN5,
        entity_category=EntityCategory.DIAGNOSTIC,
        icon="mdi:lock",
    ),

    SolaxModbusSelectEntityDescription(
        name="Manual Mode Select",
        key="manual_mode_select",
        register=0x20,
        option_dict={
            0: "Stop Charge and Discharge",
            1: "Force Charge",
            2: "Force Discharge",
        },
        allowedtypes=AC | HYBRID | GEN4 | GEN5,
        icon="mdi:dip-switch",
    ),

   SolaXModbusSensorEntityDescription(
        name="Inverter Temperature",
        key="inverter_temperature",
        native_unit_of_measurement=UnitOfTemperature.CELSIUS,
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        register=0x8,
        register_type=REG_INPUT,
        scale=0.1,
        unit=REGISTER_S16,
        allowedtypes=AC | HYBRID | GEN5,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),

   SolaXModbusSensorEntityDescription(
        name="Run Mode",
        key="run_mode",
        register=0x9,
        scale={
            0: "Waiting",
            1: "Checking",
            2: "Normal Mode",
            3: "Fault",
[... 1120 lines cut ...]
    return (
        datadict.get("inverter_power", 0)
        - datadict.get("measured_power", 0)
        + datadict.get("meter_2_measured_power", 0)
    )

def value_function_house_load_alt(initval, descr, datadict):
    return (
        datadict.get("pv_power_1", 0)
        + datadict.get("pv_power_2", 0)
        + datadict.get("pv_power_3", 0)
        - datadict.get("battery_power_charge", 0)
        - datadict.get("measured_power", 0)
        + datadict.get("meter_2_measured_power", 0)
    )

Den ons. 7. maj 2025 kl. 19.26 skrev Edwin Biemond ***@***.***

>:

>
> more like your missing load or meter params which I cannot find in your
> doc.
>
> I found temp which was also not on your pdf

biemond, 2025-05-07:
cool , let me focus on load, batt  and meter  power entries

JorgenSeemann, 2025-05-07:
Sounds good, thanks 😀

Med venlig hilsen
Jørgen

JorgenSeemann, 2025-05-11:
Confirming PV1-3 all good 😀

Med venlig hilsen
Jørgen

biemond, 2025-05-11:
nice , will see if I can add more attributes tonight.

JorgenSeemann, 2025-05-11:
Cool, let me know. 😀

Med venlig hilsen
Jørgen

biemond, 2025-05-11:
can you try the new version and send me a diagnostic report so I can check for the house load.

JorgenSeemann, 2025-05-11:
Done -> 23120829-1ca1-4971-8175-926d57ada775
:)

biemond, 2025-05-11:
checking, so you are running on your battery and your house load is 265 watt. 
hope to see some daytime with solar and changring 

plus something like stop the battery and see if grid import works.

```
Powerdc1 0 0 Powerdc1
Powerdc2 0 0 Powerdc2
Powerdc3 0 0 Powerdc2

Batpower_Charge1 -265 0 Batpower_Charge1
Batpower_Charge2 0 0 Batpower_Charge2

measured_power 0 0 measured power
meter_2_measured_power 0 0 meter 2 measured power

```

     datadict.get("pv_power_1", 0)
        + datadict.get("pv_power_2", 0)
        + datadict.get("pv_power_3", 0)
        - datadict.get("battery_power_charge", 0)
        - datadict.get("measured_power", 0)
        + datadict.get("meter_2_measured_power", 0)

JorgenSeemann, 2025-05-12:
f907bc3c-6b5b-475b-ab4d-ceab4aa75d97
PV around 1K
Load around 2K
Battery load

JorgenSeemann, 2025-05-12:
22e2f824-3965-4fb2-b0bd-84dd4ecc8054
PV -> Battery (app 1K)
Grid -> Load (App 2K)

biemond, 2025-05-12:
looks ok now 

```
    PV -> Battery (app 1K)
Grid -> load (app 2K)

GridPowerR 1356 0 Inverter Power (R)
GridPowerS -673 0 Inverter Power (S)
GridPowerT -691 0 Inverter Power (T)

Powerdc1 0 0 Powerdc1
Powerdc2 0 0 Powerdc2
Powerdc3 1324 0 Powerdc2

Batpower_Charge1 1268 0 Batpower_Charge1

BatteryCapacity 98 0 Battery Capacity

BMS_BatteryCapacity 9154 0 BMS Battery Capacity

measured_power -1946 0 measured power
meter_2_measured_power 0 0 meter 2 measured power

```

biemond, 2025-05-12:
new version https://homey.app/a/solaredge.modbus/test/

JorgenSeemann, 2025-05-12:
Updated. I now see:
Solar + Batt Power
Battery Power
PV input + PV1-3 input
Battery charge current + voltage
Total Export + import
Total Day Export + Import
Total Yield + Total Day yield
Battery Health + SOC
Heatsink temperature
Grid in/output
House load Power

I do however not see any info about grid phases.

More problematic is that the change to system run is throwing errors (eq.
trying to set to force charge, discharge etc)
The error that pop up is: Missing Capability Listener:
storage_force_charge_discharge2
If I try to set it to manual mode I get error: Missing Capability Listener:
solarcharger_use_mode
Diag is 8aba664f-7b03-4079-a3ac-869c0a948930

Cheers
Jørgen

biemond, 2025-05-12:
hey , weird I disabled those all, let me check this.

biemond, 2025-05-12:
pushed a new version

JorgenSeemann, 2025-05-12:
Ok. New version = error message is gone. But quick test doesn't seems like
it's actually changing settings. I'll try playing a bit more tomorrow and
see what happens.
Not sure if the diag is useful, but here we go
:) 844e5fb7-eab9-4b65-ba62-945c27444562

Cheers
Jørgen

biemond, 2025-05-12:
can you check if some password is needed. 

[image]

JorgenSeemann, 2025-05-13:
> can you check if some password is needed.
> Screenshot.2025-05-12.at.21.59.54.png (view on web)

Works as supposed after rebooting my Homey 😇

Med venlig hilsen
Jørgen

JorgenSeemann, 2025-05-13:
Think the house load is wrong thou.

Shows 825w, pv 823w, grid 0, battery -1227
Solar + batt 1914
House load should be around 2k as I have an electrical heater running 😉

Med venlig hilsen
Jørgen

biemond, 2025-05-13:
can you send me a report.

biemond, 2025-05-13:
so is your load around 2300?

```

Powerdc1 0 0 Powerdc1
Powerdc2 0 0 Powerdc2
Powerdc3 0 0 Powerdc2

Batpower_Charge1 -2261 0 Batpower_Charge1

measured_power -49 0 measured power

```
```
        var gridPower = Number(result['measured_power'].value) * Math.pow(10, Number(result['measured_power'].scale));

        const dcPower1 = Number(result['Powerdc1'].value) * Math.pow(10, Number(result['Powerdc1'].scale));
        const dcPower2 = Number(result['Powerdc2'].value) * Math.pow(10, Number(result['Powerdc2'].scale));
        const dcPower3 = Number(result['Powerdc3'].value) * Math.pow(10, Number(result['Powerdc3'].scale));
        var battPower  = Number(result['BatCurrent_Charge1'].value) * Math.pow(10, Number(result['BatCurrent_Charge1'].scale));

        this.setCapabilityValue('measure_power.load', Math.round(dcPower1 + dcPower2 + dcPower3 - gridPower - battPower ));
```

JorgenSeemann, 2025-05-14:
Sounds about right.

I have tried to grab the data from Solax website (below) + new debug report
: 4807beb7-1e92-4f60-bda3-e18218ca1f1d
PV Power
236W
AC Power（On-grid）
1.94kW
AC Apparent Power
1.94kW
Grid Power
0W
Other Inverter Power
0W
Power Factor
-0.020
EPS Power （Off-grid）
0W
AC Voltage (L1)
230.9V
AC Current (L1)
8.9A
AC Frequency (L1)
50.07HZ
AC Voltage (L2)
230.9V
AC Current (L2)
-1.3A
AC Frequency (L2)
50.04HZ
AC Voltage (L3)
231.3V
AC Current (L3)
-1.3A
AC Frequency (L3)
50.03HZ
Grid Voltage (L1)
231.2V
Grid Current (L1)
0.7A
Grid Voltage (L2)
231.5V
Grid Current (L2)
0.6A
Grid Voltage (L3)
230.7V
Grid Current (L3)
0.6A
EPS Voltage (L1)
0.0V
EPS Current (L1)
0.0A
EPS Frequency (L1)
0.00HZ
EPS Voltage (L2)
0.0V
EPS Current (L2)
0.0A
EPS Frequency (L2)
0.00HZ
EPS Voltage (L3)
0.0V
EPS Current (L3)
0.0A
EPS Frequency (L3)
0.00HZ
Battery Power
-1.83kW