biemond/solaredge.modbus #140: No values from Sungrow string

ostwes, 2025-04-09:
When homey energy went from beta to release i notised that my sungrow string stopped reporting values. The original app still reports values. Tried to  uninstall the app and installed the beta version. The device was found and i could connect to the inverter and add it again but the same result.

biemond, 2025-04-09:
did your ip change?  can you send me a report

ostwes, 2025-04-10:
Hi Edwin!

Sorry for reaching out to you, it works now.

I thought that i had tested everything but just realized that it could be the router that for some reason could cause this.

After restarting the router it started to report values.

The reason that i did not suspect the router was that it connected to the assigned ip.

Thank you for developing this awesome Application 🙂

Best Regards
Östen

________________________________
Från: Edwin Biemond
Skickat: Onsdag, 09 april 2025 19:45
Till: biemond/solaredge.modbus
Kopia: Östen Westermark; Author
Ämne: Re: [biemond/solaredge.modbus] No values from Sungrow string (Issue #140)

[https://avatars.githubusercontent.com/u/1085004?s=20&v=4]biemond left a comment (biemond/solaredge.modbus#140)<https://github.com/biemond/solaredge.modbus/issues/140#issuecomment-2790496339>

did your ip change? can you send me a report