biemond/solaredge.modbus #195: SolarEdge StorEdge responds slow or sometimes not

NCVervaet, 2025-05-11:
Hello Edwin, eversince about 3rd week of April, I am experiencing issues with my Homey Pro, StoreEdge inverter with home battery. Flows which used to work, fail: Commands like 'Set Storage Charge/Discharge Mode' to any of the options like 'Maximize Export' or 'Charge from PV+AC' take a lot more time to take into effect or sometimes do not respond at all. It used to take about 15-25s, but now it takes 30-60s. Some other changes used to work in the past, like from Limit Mode 'Import control' to 'Export Control'. I just found out it first needs to be set to 'Disabled' in order to make a change. 

I tried to figure out what's going on. I see my inverter's firmware has not been updated. Did something change in your app controlling SolarEdge inverters? (I see a lot of improvements for Growatt inverters when I look a the change-log)

If I know what changed for Solar Edge inverters, I can adapt my flows to incorporate for the change.

Thanks in advance,

Norbert

P.S. Can you check this issue, I think I lost connection to the inverter, which is probably my fault: f9260743-8b82-49d8-b9d9-c53332016fcf
Normally this is not the case, I have a flow that every 5 hours reconnects to the Inverter by executing 'Set Storage Control mode' to 'Remote Control'

biemond, 2025-05-11:
hey, 

weird, maybe the inverter can use some restart (waiting for that click),  but no change on solaredge for years. last change was active power limit. 

your output looks ok only see some unhandled exception but don't see any stacktrace

```
  stderr:
    unhandledRejection:
[object Object]
unhandledRejection:
[object Object]
unhandledRejection:
[object Object]
unhandledRejection:
[object Object]
unhandledRejection:
[object Object]
```

NCVervaet, 2025-06-05:
Hello Edwin,

Still got no clue, the inverter reboots a couple of times per day. I've got
a flow which reconnects every 5 hours by pressen 'Set Storage Control' to
Remote Control and shortly (90s) after that a mode, like Maximise Self
Consumption.

Right now the inverter is connected accoding to my ubiquity system, but it
does not respond to a command.
Can you please check this error message:

875cf197-b027-44b5-939b-b7482e1029a0

What mostly helps is pressing the Storage Charge/Discharge Mode a second
time in the app or in a flow. Thanks for thinking along,
Norbert

Thanks