biemond/solaredge.modbus #197: SolarEdge sunspec + Victron ESS, modbus ID issue

daanoski, 2025-05-16:
Hi,

I want to use modbus over TCP for both Homey and Victron ESS. The Homey app has an device ID limited to max 50, and Victron has an non changeable device ID of 126. Now I have to choose one or the other. Maybe someone has a solution for this issue? Thnx.

biemond, 2025-05-16:
hey,

So are you doing my homey app for solaredge or through victron as some proxy? 
I can change it to max 126

biemond, 2025-05-16:
can you try this https://homey.app/a/solaredge.modbus/test/

daanoski, 2025-05-16:
> can you try this https://homey.app/a/solaredge.modbus/test/

Thanks. I will install later today and hopefully have a successful test tomorrow

daanoski, 2025-05-16:
> > can you try this https://homey.app/a/solaredge.modbus/test/
> 
> Thanks. I will install later today and hopefully have a successful test tomorrow

I have installed the app and succeeded to connect to my solaredge inverter. It is not an storedge but I do have an solaredge grid kWh meter. My aim is to limit grid export to 0 in times of negative energy pricing. When adding the device in Homey I chose 'solaredge+storedge' because 'solaredge inverter' does not gave me the desired function. 

I have made an advanced flow and managed to enable and disable 'export limit' however the set value will always be 3000W. Even when I add 'Set export limit to' 0W. I have tested the 'grid limit' function in SolarEdge manually+successfully before. 

Many thanks again.

biemond, 2025-05-16:
nice.  yes storedge is fine, even with no battery. 

indeed 3000 is my default value , did you add a 10 second delay inbetween or retry it after x seconds.  Solaredge needs some time to process it all, you will hear a click. 

```
     if (type == 'limitcontrolmode') {
        // 0 – Disabled
        // 1 – Export Control
        // 2 – Production Control
        // 11 – Minimum Import Control
        if (value == 1) {
          const limitcontrolmodeeRes = await client.writeSingleRegister(0xe000, Number(1));
          console.log('limitcontrolmode', limitcontrolmodeeRes);
          // side = 0
          const limitcontrolsideRes = await client.writeSingleRegister(0xe001, Number(0));
          console.log('limitcontrolside', limitcontrolsideRes);
          // 3000
          const limitcontrolWattRes = await client.writeMultipleRegisters(0xe002, [32768, 17723]);
          console.log('limitcontrolwatt', limitcontrolWattRes);
```

after that do indeed export limit to 
```
{
  "title": {
    "en": "Set Export Limit to",
    "de": "Setzte Exportbeschränkung auf",
    "sv": "Ställ in exportbegränsningen till"
  },
  "args": [
    {
      "type": "dropdown",
      "name": "exportlimit",
      "values": [
        {
          "id": "0",
          "label": {
            "en": "0W"
          }
        },

```

daanoski, 2025-05-17:
> nice. yes storedge is fine, even with no battery.
> 
> indeed 3000 is my default value , did you add a 10 second delay inbetween or retry it after x seconds. Solaredge needs some time to process it all, you will hear a click.
> 

It works! Thank you.