biemond/solaredge.modbus #199: Sungrow and managing the battery

bergh-io, 2025-05-27:
Apparently I no longer can set the battery to self-consumtion or Forced mode via a flow. I can set it via the menu in the device.
I cant, either via flow och menu, set the charge/discharge command. If i use the webinterface i get the error "not_setable"

I made a ticket after I got the error in the webinterface.
f354f353-6ac0-4981-8187-64f65fbb92f9 

Please let me know if I can provide any more insight to help solve this as I now can´t manage the system and optimizing my setup.

The readback values are are updated as it should so it just the commands that no longer work.

bergh-io, 2025-05-27:
I have two devices, and come to think of it I only had the first one when trying it out earlier. 

It would be awesome if you find a solution :)

biemond, 2025-05-27:
yeah will fix it and know what it is . 

I see 4 devices,  for now keep '192.168.1.28', and '192.168.1.27',

remove 

```
Error: connect EHOSTUNREACH [192.168.1.48:516](http://192.168.1.48:516/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1157:16) {
  errno: -113,
  code: 'EHOSTUNREACH',
  syscall: 'connect',
  address: '192.168.1.48',
  port: 516
}
Client closed
Error: connect EHOSTUNREACH [192.168.1.64:502](http://192.168.1.64:502/)
    at TCPConnectWrap.afterConnect [as oncomplete]
```

bergh-io, 2025-05-30:
Thanks!

It seems to be working for setting everything as intended with flows :)

Only thing I can see that do not seem to work is setting the Charge/discharge/stop via the menu i the device. But it does works via flow. 
I created a log for it if that helps: 1f631495-2cb7-42d0-9ee5-9d3b8f65da98

My main ting is via flow and that works perfectly now, thank you!