biemond/solaredge.modbus #203: Huawei THEN actions

weide43, 2025-06-17:
I can do actions manually on my Huwawei battery (Huawei Inverter + Battery) but in the flow cards I got only WHEN cards and nog THEN cards.
Because I want to charge the battery at night with negative prices.

biemond, 2025-06-17:
you mean I don't see then actions or they do not work? 

are you using s-dongle?    else see this https://github.com/biemond/solaredge.modbus/issues/89#issuecomment-2979835555

weide43, 2025-06-17:
> you mean I don't see then actions or they do not work?
> 
> are you using s-dongle? else see this [#89 (comment)](https://github.com/biemond/solaredge.modbus/issues/89#issuecomment-2979835555)
I mean, I don't see actions only WHEN cards but no AND/THEN cards

Yes I use a S-dongle

closed 2025-06-17 by weide43