biemond/solaredge.modbus #204: Not getting value anymore from Huawei Inverter alone

mercierv, 2025-06-25:
Hi :) 

Since 3 days, I'm not getting any value anymore from my Huawei inverter. 
I've created a log report: b14075e5-f1cc-40f5-bfe7-8bee4f7fddb1 

The report shows that there is not connection anymore between the Inverter and Homey. 
Which make sense as I can't ping the inverter; neither a test-netconnection on the port 502 
I've got a S-Dongle as well, which I can ping without any issue. I don't really understand why suddenly it's not working anymore as I didn't changed anything. 
Maybe someone could help me on this ? 

For your information, I've setup the device in Homey as the IP of the inverter not the S-Dongle. 
Printscreen of last update:

[image]

I don't have any firewall put in place, but the S-Dongle is linked to a Powerline which is link to a mesh network, if that can help. 

Cheers,

biemond, 2025-06-25:
hey, no clue but I think I need to talk to the s-dongle , that is the master modbus.   but you need to check this all.  there are some home assistant and other docs how to do modbus tcp.

mercierv, 2025-06-25:
Even with the s-dongle I receive no data, I´ve tried both. 
modbus tcp was activated on my appliances by the provider, I will check with them as well if an update on their side may have disconnected the modbus. 
Cheers