biemond/solaredge.modbus #205: Battery level and charging state not reacting

Xybertecnic, 2025-07-03:
I'm trying to use the level of the batteries connected to my Growatt SPA10000TL3 to send commands to my solar panels (SolarEdge), but for some reason nothing is working, even though I see the battery level and the charging/discharging state change in the device itself (in homey), anyone an idea why these triggers wouldn't work?

[image]

[image]

biemond, 2025-07-03:
need to check I think they come from homey itself.   are you using the test version of the app?

Xybertecnic, 2025-07-03:
Hallo Edwin, ja ik gebruik de test app

biemond, 2025-07-03:
will check what it can be.

Xybertecnic, 2025-07-03:
It works now, but the only card that reacts a trigger is 

[image]

All the others don't do anything

biemond, 2025-07-03:
will check

biemond, 2025-07-04:
so this works then https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/flow/triggers/meter_power_changed.json

and this not https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/flow/triggers/measure_power_battery_charge_changed.json

https://github.com/biemond/solaredge.modbus/blob/main/.homeycompose/flow/triggers/measure_power_battery_discharge_changed.json

biemond, 2025-07-04:
ack , will check from solaredge batt (dis)charge