biemond/solaredge.modbus #206: Support for displaying battery as energy source in Homey Energy (Virtual House view)

Hurbie412, 2025-07-03:
Summary:
The app successfully exposes battery power data (e.g., measure_power.batt_charge_discharge) via Modbus. However, the battery is not shown as an energy source in the Homey Energy dashboard ("Virtual House" top section). I would like to request support for this feature.

Current Behavior:
Battery appears either not at all, or only as a device with consumption.
The actual power data is visible under device capabilities and updates correctly.

Expected Behavior:
Battery (e.g. Growatt / Sigen) should be shown in the top section of the Homey Energy view, alongside other sources like solar panels.

Technical details:
To enable this, the app would likely need to set the following in the driver definition:

json
Kopiëren
Bewerken
"energy": {
  "energy_source": true
}

This tells Homey that the device contributes power to the household, allowing it to be visualized correctly in the Energy UI.

Why is this important?
The battery is a critical component in managing household energy.
Users expect to monitor charge/discharge like solar panels.
Current presentation in Homey doesn't reflect its real function in the system.
Suggested implementation:
Add energy_source: true to the battery driver
Bind measure_power.batt_charge_discharge or equivalent to measure_power
Optionally: differentiate between charging (+) and discharging (-)

System setup:
Homey Pro [model/year]
Solaredge + Growatt TCP Modbus app (latest version)
Growatt / Sigen battery connected via Modbus TCP

Let me know if I can help test this or provide more data.

Thanks in advance!

biemond, 2025-07-03:
will verify  but probably not.  for sure I will not make batt devices , homey has to solve this.

you need to make a new device with  device class as battery and set the homeBattery property in the energy object of the driver.
```
  "energy": {
    "homeBattery": true,
    "meterPowerImportedCapability": "meter_power.charged",
    "meterPowerExportedCapability": "meter_power.discharged"
  }
```

https://apps.developer.homey.app/the-basics/devices/energy#home-batteries

biemond, 2025-07-03:
ask them to make a new device type like solar with battery, that has  solar, meters and battery.  so I can provide all the fields and then you don't need to have 3 devices, with modbus you cannot have 3 devices , mostly 1 keeps the lock.

biemond, 2025-07-09:
awesome