biemond/solaredge.modbus #210: Sigenergy: THEN card does not work

softronicjohan, 2025-08-06:
Hi!

After the major update, the "Remote EMS" card no longer works for Sigenergy cards.

However, I can still control the Sigenergy system by manually activating EMS Mode from Homey via the device, so it seems the issue is with the card itself.

biemond, 2025-08-06:
ack, do you have the release notes and new modbus register doc for me?

PressRec, 2025-08-13:
Found a 2.7 version. have you had any time looking into it?

[Modbus Protocol EN - SIGEN - V2.7 250523.pdf](https://github.com/user-attachments/files/21748242/Modbus.Protocol.EN.-.SIGEN.-.V2.7.250523.pdf)

biemond, 2025-08-13:
oh, must be simple then.

biemond, 2025-08-13:
found it , can you try this https://homey.app/a/solaredge.modbus/test/