biemond/solaredge.modbus #214: Sigenergy - Energy tab, solar/battery

Wiibeerg, 2025-09-25:
Hi

Thank you for this app.
This is more of a question than an issue.

For the Energy tab "Solar+battery" is used to generate data in Solar for the Homey Energy tab. But this is not displayen correct if the bettery is charged from the grid. - Is there a reason to why "Solar+battery" is used instead of "PV input"? 
- Can I change this somehow?
- Is there a plan to seperate solar and battery? 

It easy to make your own dashboard to display this, why do Homey make it so hard to get i to display correctly in the Energy tab? 

Best regards
Johan

Wiibeerg, 2025-09-25:
Ok, thanks for the quick reply!

I guess then the only way to make it look correct is to have the battery visible. Hope homey fix it within a reasonable time.

closed 2025-09-25 by Wiibeerg

biemond, 2025-09-25:
they are aware , please respond to them to give it a higher prio.

Wiibeerg, 2025-09-25:
I will do that.

The summary in kWh is missing for solar panel production. Is that also related to the things Homey need to fix or is that data missing from Sigenergys modbus?

biemond, 2025-09-26:
I guess it is not there, I only remember daily ones. need to check the docs.