biemond/solaredge.modbus #215: Sigenergy - "Total day"-data is missing

Wiibeerg, 2025-09-28:
Hi

The following data field is present in the Homey app, but it is always zero:
- Total day import

The following data is present in the Sigenergy app, but not in the Homey app:
- Total day PV input (Solar produced) 
- Total day Consumed Power (Load consumed)

Not sure if this is related, but my understanding is that because of this there is no day total data history in the Energy tab. 

[image]

biemond, 2025-09-28:
Hey, 

can you send me a report. 

for energy tab these are important.
```
    "sigen_plant_pv_total_generation":    [30088, 4, 'UINT64', "Plant PV total generation", -2],
    "sigen_total_imported_energy":        [30216, 4, 'UINT64', "Total imported energy", -2],
    "sigen_total_exported_energy":        [30220, 4, 'UINT64', "Total exported energy", -2],

```

Wiibeerg, 2025-09-28:
Diagnostics Reporte: 2f0b9b3-ef55-4e8e-9b6c-529f043f3a28

biemond, 2025-09-28:
can you send me another 1 then

biemond, 2025-09-28:
still not

you have some other app running which blocks 502 port 

Error: connect ECONNREFUSED [10.10.1.113:502](http://10.10.1.113:502/)
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '10.10.1.113',
  port: 502

Wiibeerg, 2025-09-28:
Not sure why 10.10.1.113 is show in the error log.
The settings for the device is set to 10.10.1.209.

biemond, 2025-09-28:
looks way better now,  looks I like I am missing half of it. 

```
sigenergy:  sigendev_daily_export_energy 480 -2 Sigen Daily export energy
sigenergy:  sigendev_accumulated_export_energy 5799 -2 Sigen Accumulated export energy
sigenergy:  sigendev_daily_import_energy 0 -2 Sigen daily import energy
sigenergy:  sigendev_accumulated_import_energy 7 -2 Sigen Accumulated import energy
sigenergy:  sigendev_daily_charge_energy 158 -2 Sigen Daily charge energy
sigenergy:  sigendev_daily_discharge_energy 102 -2 Sigen Daily discharge energy

```

biemond, 2025-09-28:
I am missing a lot of registers, can you do an update of the firmware and maybe reboot inverter etc and restart homey app.

Wiibeerg, 2025-09-28:
Restarted inverter, restarted Homey and restarted the app.
11bfae86-3ecf-4244-91eb-7b55b5f9c70d

Maybe need to reinstalled the app? But then I lose all insight data?

Wiibeerg, 2025-09-29:
Thank you, now I can see the total solar production per day.
"Total day import" is still 0. But Maybe I'm interpreting it incoerrectly, it might be only import from grid to battery and not from grid to consumption? 

A nitpick is that I had a hard time finding the correct match between Power and Energy.
Is it possible to change the names so the Power(W) is easy to match with the corresponding Energy(kWh)

(Used Homey Energy dongle as naming reference) 

Example Power
Power *type*
Power Load
Power Solar
Power Grid
Power Battery

Example Energy
Energy *type* Today
Energy Load Today
Energy Solar Today
Energy Grid Import Today
Energy Grid Export Today
Energy Battery Charge Today
Energy Battery Discharge Today

Example Total Energy
Total Energy *type*
Total Energy Load
Total Energy Solar
....

biemond, 2025-10-01:
will check the docs but has nothing to do with battery ,  are you normally doing  some importing?

charging is this sigendev_daily_charge_energy 158 -2 Sigen Daily charge energy

```

    sigendev_daily_import_energy: [30560, 2, 'UINT32', 'Sigen daily import energy', -2],
    // precision: 2
    // unit_of_measurement: kWh
    // scale: 0.01
```

Wiibeerg, 2025-10-01:
Total day Import seems to representant this:
[image]

By they name I would assume it would fetch this:
[image]